bigbigmdm/IMSProg #4: Help with settings for S93C56 BDDVAX 8913

luther7, 2023-09-20:
I'm trying to read a chip marked S93C56 BDDVAX 8913. Any idea what settings I should use? Thanks!

luther7, 2023-09-20:
Thanks! I think it may be my adapter.

[image]
[image]

bigbigmdm, 2023-09-20:
I testing only 16bit mode.

bigbigmdm, 2023-09-20:
Can you please tell me where I can buy such an adapter?

luther7, 2023-09-20:
> Can you please tell me where I can buy such an adapter?

I purchased it from here: https://www.ebay.com.au/itm/195699712014

luther7, 2023-09-20:
With a jumper across R4 in the converter, I think I can read it correctly using SNANDer.

luther7, 2023-09-20:
Thanks for your help @bigbigmdm :heart:

closed 2023-09-20 by luther7