bigbigmdm/IMSProg #5: Bios Chip MX25L12835EZNI-10G

trinityprivacy, 2023-10-10:
Hi,

From [this](https://www.reddit.com/r/coreboot/comments/14xizim/difficulty_with_green_ch341a_v17/) discussion I have this problem:

When I try to flash coreboot.rom file but I get this error when I do the "Erase-Program-Verify":

Error comparing data!
Address: 00000000
Buffer: 5F
Chip: 00

My bios chip is MX25L12835EZNI-10G. Is that supported?

bigbigmdm, 2023-10-10:
Hi!
This chip is supported. According to JEDEC information and parameters, this chip is equivalent to MX25L12805D. I see two problems:
1. It is possible that the chip is damaged.
2. Perhaps the chip is hardware locked by the third status register and secured OTP. (Please see the page number 51 of the [datasheet](https://www.alldatasheet.com/datasheet-pdf/pdf/934247/MCNIX/MX25L12835EZNI-10G.html)  ) 
In both cases, it will not be possible to write such a chip.
The manufacturer of the chip is MXIC company. Yesterday I tested the MX25L1006BE chip from this manufacturer. Everything was in perfect order.
If you have another programmer, please test this chip and write me the result.

trinityprivacy, 2023-10-10:
Thanks for your reply!

Before trying this external flashing, I first tried internal flashing, ad documentate on coreboot website and [here](https://review.coreboot.org/plugins/gitiles/coreboot/+/refs/changes/66/36666/1/Documentation/mainboard/lenovo/ivb_internal_flashing.md#examining-protections-theory). And also I remember that WP was set to 0

After flashing, something must have gone wrong and my T430s started but the screen was black. That's why I opened it and tried to flash it with ch341a 1.7

With flashrom I tried but nothing works.
I will try with the other 2 programs you mentioned.

My equipment is just ch341a 1.7 and a pogo pin adapter for WSON8 chip (8x6)

bigbigmdm, 2023-10-10:
If all software does not program this chip, you can use a [new (not locked)](https://aliexpress.ru/item/1005002625935964.html) chip, and program it with any software.

trinityprivacy, 2023-10-10:
I used AsProgrammer and NeoProgrammer on VM as well (I have Linux system). It seems they do something, as the led of the ch341a is green during the read for example. I'll explore if there is a specific option to set for OTP.

Regarding the chip I do not have desoldering experience but if I cannot solve my problem I'll have to consider this option.

I'll keep you updated!

bigbigmdm, 2023-10-10:
Thank you, I look forward to the sequel....

bigbigmdm, 2023-10-10:
Only now I realized that you are using the "clothespin" to connect to the chip.  When using the "clothespin", there is no guarantee that the chip will be read and written correctly. As a last resort, you should cut the power track (connected to pin 8) .

trinityprivacy, 2023-10-11:
This is my equipment:

[image]

I also added shorter cables as you can see.

Could you explain in detail what I have to do as cutting the power track?

Thanks :)

bigbigmdm, 2023-10-11:
You understand that you have connected the CH341a programmer to the SPI NOR flash bios chip, but the SPI NOR flash bios chip is already connected to the microprocessor, and the power pin is already connected to the RAM, buffers, north and south bridges and other devices on the motherboard? As a result, the programmer sends signals to all connected devices. Such connection does not guarantee correct execution of read and write operations.

bigbigmdm, 2023-10-17:
Please download the latest version of the program and run it. Then press <Ctrl+P> and send me a screenshot of the result.
[image]

trinityprivacy, 2023-10-19:
So, after installing the new version, connecting to the bios chip, and pressing CTRL+P, I get different values every time I press the button `Read` :(

bigbigmdm, 2023-10-19:
This proves - you can't use the clip in this motherboard! If a clip is used, the data is distorted.

trinityprivacy, 2023-10-19:
Oh no :( ! So the only way is to desolder the chip?

bigbigmdm, 2023-10-19:
Yes, desoldering is required. I'm so sorry...

trinityprivacy, 2023-10-19:
No worries man, you have been so kind and I appreciate a lot your help. It is a endless journey, I am getting crazy lol!
I have never desoldered, so I'll see on how to do it

closed 2023-10-26 by bigbigmdm

trinityprivacy, 2023-11-06:
Can you recommend a cheap desoldering kit? Or a specific one is needed?

bigbigmdm, 2023-11-06:
In my home, I use (this)[https://aliexpress.ru/item/33042036119.html] soldering station.