bigbigmdm/IMSProg #7: Feature: recent files

carbofos, 2023-10-26:
Thank you for a good software but the feature is realy missing is recently open files.
Also it does not  remember last opened directory. All the path should be passed again and again from the start to open a firmware.

bigbigmdm, 2023-10-26:
Thanks for the good review of IMSProg. I will think about the functions to memorise file paths.

bigbigmdm, 2023-10-27:
Your wish has come true!
Ver. 1.0.28 - Bad display of qComboBoxes in UBUNTU, Manjaro linux fixed, display of monospaced font in Manjaro linux fixed. Changed the default file path to the home directory. The last file path is saved and new file operations are processed using the last path.

carbofos, 2023-10-27:
Thank you.
I did check out the last version and it's more beatuful.
But if I close the program and run it again it does not remember the last directory again.
It is not a big problem but it's annoying when the path is very long.

bigbigmdm, 2023-10-28:
When I read your comment, I thought that this function is exactly as I made it. I did not plan to create an .ini file and store the last path in that file.

carbofos, 2023-10-30:
I do not know why you do not want to save current settings in .ini file but it's exactly the same way like all popular software works. You can check out the behaviour. E.g. vscode, LibreOffice, NeoProgrammer e.t.c.

bigbigmdm, 2023-10-31:
You can add a symbol link to the EEPROM directory that will be displayed when you open the file in the IMSProg program.
`ln -s /home/<...your folder path...> MY_EEPROM`.
[image]

 I will probably add the ini file to IMSProg. But that's not the first priority. Right now I need help in creating a DEB package with source code to be hosted on UBUNTU Launchpan.net.

closed 2023-11-09 by bigbigmdm