bigbigmdm/IMSProg #9: Not found file `IMSProg_editor` (dep pkg)

Podolyakofs, 2023-11-09:
Good day, 
after installing deb package (v1.0.29-3/Ubuntu 22.04)  can't open "Edit chip database" (ctrl+q) with "Not found file `IMSProg_editor` error" and same error on program start

[image]

Separately IMSProg_editor can be opened
[image]

bigbigmdm, 2023-11-09:
Package version 1.0.29-3 does not exist. Please install version 1.0.29-2 and test it. The /opt/IMSProg folder should contain the following files: IMSProg, IMSProg.Dat, IMSProg_editor.

Podolyakofs, 2023-11-09:
Thanks, downgraded package - now everything works

closed 2023-11-09 by Podolyakofs

bigbigmdm, 2023-11-19:
Can you please tell me if this is the main branch? I run this program normally...