bigbigmdm/IMSProg #16: Source for chip records update

blackPantherOS, 2023-11-15:
Hi again,

I would like to recommend a source for updating the IMSprog database, a good source the mtd drivers (kernel/drivers/mtd/devices/) from kernel tree, example:
https://www.infradead.org/pipermail/linux-mtd/2013-April/046407.html

bigbigmdm, 2023-11-15:
It is very good! Thank you!

bigbigmdm, 2023-11-17:
I added the chips from your link to the IMSProg database file.

blackPantherOS, 2023-11-19:
> I added the chips from your link to the IMSProg database file.

Nice! I am also updating the package now on our distribution.

blackPantherOS, 2023-11-19:
I found a chip with the manufacturer GallopMem and the identifier 25Q64ASIG in an Anyka IP Cam. I couldn't find a datasheet specifically for this, but it seems that I can read/write it as BY25Q64A.
[image]

bigbigmdm, 2023-11-19:
Some small manufacturers copy the entire chip from others along with the JEDEC ID. At the same time, all parameters of such chips are also the same.

bigbigmdm, 2024-01-16:
The [TL866 chip database](https://gitlab.com/DavidGriffith/minipro/-/blob/master/infoic.xml?ref_type=heads) can be used to search for new chips.

aluks01, 2024-01-31:
Hi if I wanted to write file to the MICROCHIP 25AA1024 chip could I add this to the chip database ? Currently I can't get the program to detect the chip.[]((https://ww1.microchip.com/downloads/aemDocuments/documents/MPD/ProductDocuments/DataSheets/1-Mbit-SPI-Bus-Serial-EEPROM-Data-Sheet-20001836K.pdf))

bigbigmdm, 2024-02-01:
@aluks01, it is SPI, but not NOR flash. I plan to add this class of chips. Today, use `95_EEPROM` -> `ST` -> `M95M01`. This is the analogue of your chip.

aluks01, 2024-02-01:
> @aluks01, it is SPI, but not NOR flash. I plan to add this class of chips. Today, use `95_EEPROM` -> `ST` -> `M95M01`. This is the analogue of your chip.

@bigbigmdm  thanks for sharing much appreciated !

bigbigmdm, 2024-02-11:
@aluks01 Tomorrow I will be testing a new version with support for the 25XXX chip.

aluks01, 2024-02-11:
@bigbigmdm  thanks for the update. I managed to flash the 25AA1024 by selecting 95_EEPROM -> ST -> M95M01, but it's good to know in future there will be support for 25XXX chips adding more functionality.

bigbigmdm, 2024-02-12:
@aluks01 , it's ready!

LtqxWYEG, 2025-06-16:
Shouldn't any 25xxx be under "25_EEPROM"? I was wondering why this section is so empty. I also didn't find the manufacturer ST, but found that "MICRON" is having identical JEDEC IDs in your list. 
I Couldn't find my 25P05 chip in your database, saw that the 25_EEPROM section has only 4 manufacturer and gave up on your software, thinking the database is just really empty. However ...

Today I was in the mood of tinkering and had an issue with AsProgrammer, where a 25P05 has been reading a different set of clear text every time I read it, as if a function was called, and tried your program again. It was still version 1.5.3s database, so I updated it with the newest, thinking maybe now it will be detected. It wasn't. So I was about to give up on IMSProg again, but felt tinkery.
Looking manually, after the DB update, I ACCIDENTIALLY selected the "SPI FLASH" list, which is where I found a whole lot more than just 4 manufacturers. Thinking "hey, the new database is much better" I was looking through it for ST manufacturer but still couldn't find it. So I was thinking, "WTH? Why would there be no ST at all?" and looked into the database myself for anything close to "25p05" and its JEDEC ID and found that they are under "MICRON" for some reason, but just stop at 128K so that my chips weren't found. The 25P05 is the 64K variant of the 25P10

Reading through this issue, I can see that your database has a lot of problems. My own problems with it are proof of it. Is Micron really bigger than ST or are their chips more prevalent? What is going on here that you just don't have ST in your DB?

BTW, ST has A versions of the 25P05 and 25P10 with bigger page size. (You can find them in both AsProgrammers and the Original CH341s programmers DB.) Both of those are not in your DB:
```
      <ST25P05 id="202010" page="128" size="65536"/>
      <ST25P05A id="202010" page="256" size="65536"/>
      <ST25P10 id="202011" page="128" size="131072"/>
      <ST25P10A id="202011" page="256" size="131072"/>
```
My version of the 25P05 is an ST 25P05VP where I couldn't find a datasheet.  (Found at least a mention of the VP variant here, but no datasheet: https://datasheet.es/stock/price.php?item=25PO5VP) Seems they are 25P05 equivalents, not A, so max page size 128 when writing. I assume that is the "sector size" in your DB editor? Another issue! The only mention of "sectors" in the datasheet is here, and there are only two in the whole chip: "The memory is organized as 2 sectors, each containing 256 pages".  
The rst in that section reads: "Each page is 128 bytes wide. Thus, the whole memory can be viewed as consisting of 512 pages, or 65536 bytes." So, in your editor there is no "page size" even though it is important for it to be 128, not 256.

You know, you are allowed to just go through other DB and copy entries. (Or the sorting/naming scheme) Why didn't you do that?

Can your program handle duplicate JEDEC IDs in the DB? No it can't: There is no pop-up asking you to choose between variants of the same JEDEC ID.  
Then please add the non A variants of the 25P05 and 10 and maybe think about adding a whole 25xxx ST section.  
Actually, I did that to my DB and tried copying the new entries here for easy 5 sec implementation if you read my comment. But looking at the file in notepad, it seems to just not include the JEDEC ID, "sector"-size or anything else I can add in your editor.  
Seeing that, it would maybe just be better to redo the whole thing as a normal xml, like both the original CH341 programmer and AsProgrammer uses. Just much easier. (But that explains why you don't just go there and copy entries. You can't, without much effort)

You know, I make this a new issue. It's too bad and unreasonable. It's a bug! xD