bigbigmdm/IMSProg #17: Editor launch from toolbar

blackPantherOS, 2023-11-19:
To run the editor program from the main window of the programmer: It seems that you didn't apply[ the my patch to the editor's relative path in mainwindow.cpp](https://github.com/blackPantherOS/IMSProg/commit/0bf6b2795d010781334dbd84def8c870bdbad125), so the program can't be launched from the programmer's toolbar, only from the command line or a separated icon. The reason for this is that when the user launches itt, it is not in /usr/bin, but in its home directory, therefore the relative path will be invalid.

I am attaching two pictures to help you understand the problems...

**First showing the binary compiled from the current source code:** 
[image]

**And this binary working fine after applying the my patch :**
[image]

bigbigmdm, 2023-11-19:
Can you please tell me if this is the main branch (1.1.2 tag)? I run this program normally...

bigbigmdm, 2023-11-19:
Please show me that path again :)

bigbigmdm, 2023-11-19:
I'm sorry! I found this Pull Request!

bigbigmdm, 2023-11-20:
Please see the latest changes to the source code.

bigbigmdm, 2023-12-25:
Thank you. I link to you in the debian/copyright.

closed 2023-12-25 by bigbigmdm