bigbigmdm/IMSProg #20: Скорость записи.

DJs3000, 2023-11-22:
Михаил, ещё раз приветствую. Заметил забавную особенность в плане разной скорости записи одной и той же флешки разным софтом. Шью флешку SPANSION S25FL064A на линуксе при помощи вашего IMSProg и на винде AsProgrammer https://github.com/nofeletru/UsbAsp-flash
Разница в скорости записи в 10 раз т.е. запись 5 мегабайтной прошивки в IMSProg проходит за ~20минут, а в AsProgrammer за ~2минуты. Нельзя ли как-то ускорить запись и в IMSProg
[image]
?)
[image]

bigbigmdm, 2023-11-22:
Это странно. На [4PDA](https://4pda.to/forum/index.php?showtopic=1077823&st=20) мне сказали, что IMSProg быстрее, чем NEOProgrammer.  Про ASProgrammer не знаю. Winbond W25Q32BV (половинка от объема Вашей микросхемы) у меня считывается за 34сек, программируется за 5 мин 23 сек. В любом случае посмотрю, но не прямо сейчас. Сейчас первоочередная задача выложить IMSProg в репозиторий UBUNTU, а с этим пока засада. Потом займусь оптимизацией скорости. Дальше мечтаю перейти к CH347 - в консольной утилите этот программатор читает 25Q32 (4Mb) за 2 секунды.

DJs3000, 2023-11-22:
Но могу зато IMSProg попадёт в репазиторий Эльбрус ОС, я его туда внесу.

bigbigmdm, 2023-11-22:
Буду премного благодарен!

DJs3000, 2023-11-22:
Видимо мне сбойная машина попалась. Вот сейчас прошивали биос с другой машины такую же флешку и со скоростью было высё хорошо.

closed 2023-11-24 by DJs3000