bigbigmdm/IMSProg #21: Обновление базы поддерживаемых микросзем.

DJs3000, 2023-11-24:
Если установить IMSprog, то в апликейшен попадает 3 файлика.
1. Программа IMSprog
2. Обновление базы поддерживаемых микросхем
3. Редактирование базы поддерживаемых микросхем
На мой дилетантский взгляд не совсем удобно получилось. По скрину видно, что 1 и 2 пункт названы одинаково, что создает путаницу. Может быть есть смысл обозвать этот пункт как-то так: IMSprog base update? Если совсем начать умничать, то можно редактор и обновление поместить в основную программу или ссылки на эти программы были в основной меню IMSprog дабы не плодить 3 ярлыка. Возможно так будет удобнее.

[image]

bigbigmdm, 2023-11-24:
Понял, поменяю desktop файл, возможно на IMSProg-updater - так покороче. Просто у меня две строки на каждый пункт, поэтому не обратил внимания. 
Не закрывайте пока, пожалуйста, а то я могу забыть...
[image]

bigbigmdm, 2023-11-24:
Поменял на Name=IMSProg_updater - можно пересобрать.

DJs3000, 2023-11-24:
Отлично! У тебя тут есть комментарий, а в xfce у меня не показывает этого.

closed 2023-11-24 by DJs3000