bigbigmdm/IMSProg #24: Issue with FairChild 93C56

Calosha, 2023-12-13:
Hi @bigbigmdm! First of all thank you for the software. I read content of 93C56 and bits are misplaced. 
I got:
`4a31 4634 3446 5338 5958 324c 3036 3333 3135`....
instead of:
`314a 3446 4634 3853 5859 4c32 3630 3333 3531`....

Any ideas what could cause incorrect reading? 

[Here](https://html.alldatasheet.com/html-pdf/51970/FAIRCHILD/FM93C56EM8/1237/3/FM93C56EM8.html) is data spec for the eeprom.
Image of the chip:
[image]

Chip connection:
| CH341A | 93C56 pin name | 93C56 pin |
|--------|----------------|-----------|
| CLK    | sk             | 2         |
| CS     | cs             | 1         |
| MOSI   | di             | 3         |
| MIOS   | do             | 4         |
| GND    | gnd            | 5         |
| 3.3V   | pe             | 6         |
| 5V     | pre, Vcc       | 7,8       |

Calosha, 2023-12-13:
I think the issue is endianness (Little Endian/Big Endian). Is there a setting for it in the software?

bigbigmdm, 2023-12-13:
There is a lot of confusion with the 93Cxx chips in some programmes. These are 16-bit chips. The high byte is read first, then the low byte. And in the 8-bit dump - vice versa. And the dump of the IMSProg programme is 8-bit. In the source code of SNANDer, which I took as a basis, there is just confusion with the byte order. I rearranged them in the correct order, just like in the EZP2019 programme. 
You are just showing me a 16-bit dump and in it the bytes should be rearranged compared to the 8-bit dump. The data will be written to the disc in a different order.

16 bit DUMP                            8 bit DUMP
4a31 4634 3446 5338 5958 = 31 4a 34 46 46 34 38 53 58 59

Am I making myself very unclear?

Calosha, 2023-12-13:
Спасибо Михаил. Тогда можно попросить фичу что-бы editor мог переключаться с 8 на 16 и обратно?

bigbigmdm, 2023-12-14:
К сожалению, быстро это не сделать, поскольку я использую для отображения дампа готовый компонент [QHexEdit2](https://github.com/Simsys/qhexedit2), а у него нет 16-битного режима. Буду думать в этом направлении, поскольку это правильно.

bigbigmdm, 2023-12-25:
Пока не реализовал, но помню о Вас.

closed 2023-12-25 by bigbigmdm

Calosha, 2023-12-25:
Спасибо! С праздниками Вас!