bigbigmdm/IMSProg #26: Wrong udev rules installation from .deb

mrv96, 2023-12-20:
It is not 100% clear where 99-CH341.rules should be placed in:

- `/lib/udev/rules.d` according this: https://github.com/bigbigmdm/IMSProg/blob/5f5b62559de889065f2ee97ee42f525e440a7e87/README.md?plain=1#L273
- `/etc/udev/rules.d` according this: https://github.com/bigbigmdm/IMSProg/blob/5f5b62559de889065f2ee97ee42f525e440a7e87/IMSProg_programmer/other/99-CH341.rules#L1

Anyway .deb actually install it in `/usr/lib/x86_64-linux-gnu/udev/rules.d/` which is surely wrong:
```bash
$ dpkg -c imsprog_1.1.2-12_amd64.deb 
drwxr-xr-x root/root         0 2023-12-01 00:08 ./
drwxr-xr-x root/root         0 2023-12-01 00:08 ./etc/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./etc/imsprog/
-rw-r--r-- root/root     36448 2023-12-01 00:08 ./etc/imsprog/IMSProg.Dat
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/bin/
-rwxr-xr-x root/root    580152 2023-12-01 00:08 ./usr/bin/IMSProg
-rwxr-xr-x root/root       549 2023-12-01 00:08 ./usr/bin/IMSProg_database_update
-rwxr-xr-x root/root    129568 2023-12-01 00:08 ./usr/bin/IMSProg_editor
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/lib/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/lib/x86_64-linux-gnu/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/lib/x86_64-linux-gnu/udev/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/lib/x86_64-linux-gnu/udev/rules.d/
-rw-r--r-- root/root       248 2023-11-17 18:17 ./usr/lib/x86_64-linux-gnu/udev/rules.d/99-CH341.rules
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/applications/
-rw-r--r-- root/root      2094 2023-12-01 00:08 ./usr/share/applications/IMSProg.desktop
-rw-r--r-- root/root      2106 2023-12-01 00:08 ./usr/share/applications/IMSProg_database_update.desktop
-rw-r--r-- root/root      1873 2023-12-01 00:08 ./usr/share/applications/IMSProg_editor.desktop
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/doc/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/doc/imsprog/
-rw-r--r-- root/root      1342 2023-12-01 00:08 ./usr/share/doc/imsprog/changelog.Debian.gz
-rw-r--r-- root/root     11863 2023-12-01 00:08 ./usr/share/doc/imsprog/copyright
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/doc/imsprog/html/
-rw-r--r-- root/root    675023 2023-12-01 00:08 ./usr/share/doc/imsprog/html/index.html
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/doc-base/
-rw-r--r-- root/root       268 2023-12-01 00:08 ./usr/share/doc-base/imsprog
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/imsprog/
-rw-r--r-- root/root      3823 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_de_DE.qm
-rw-r--r-- root/root      3771 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_es_ES.qm
-rw-r--r-- root/root      3778 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_hu_HU.qm
-rw-r--r-- root/root      3732 2023-12-01 00:08 ./usr/share/imsprog/chipEditor_it_IT.qm
-rw-r--r-- root/root      3791 2023-12-01 00:08 ./usr/share/imsprog/chipEditor_pt_BR.qm
-rw-r--r-- root/root      3782 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_ru_RU.qm
-rw-r--r-- root/root      3782 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_uk_UA.qm
-rw-r--r-- root/root      2938 2023-11-17 18:17 ./usr/share/imsprog/chipEditor_zn_CH.qm
-rw-r--r-- root/root     18193 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_de_DE.qm
-rw-r--r-- root/root     18500 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_es_ES.qm
-rw-r--r-- root/root     18628 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_hu_HU.qm
-rw-r--r-- root/root     17847 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_it_IT.qm
-rw-r--r-- root/root     18567 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_pt_BR.qm
-rw-r--r-- root/root     17894 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_ru_RU.qm
-rw-r--r-- root/root     18436 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_uk_UA.qm
-rw-r--r-- root/root     14767 2023-12-01 00:08 ./usr/share/imsprog/chipProgrammer_zn_CH.qm
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/man/
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/man/man1/
-rw-r--r-- root/root       755 2023-12-01 00:08 ./usr/share/man/man1/IMSProg.1.gz
-rw-r--r-- root/root       885 2023-12-01 00:08 ./usr/share/man/man1/IMSProg_database_update.1.gz
-rw-r--r-- root/root       793 2023-12-01 00:08 ./usr/share/man/man1/IMSProg_editor.1.gz
drwxr-xr-x root/root         0 2023-12-01 00:08 ./usr/share/pixmaps/
-rw-r--r-- root/root      6976 2023-11-17 18:17 ./usr/share/pixmaps/IMSProg64.png
-rw-r--r-- root/root      3013 2023-11-17 18:17 ./usr/share/pixmaps/IMSProg_database_update.png
-rw-r--r-- root/root      2790 2023-11-17 18:17 ./usr/share/pixmaps/chipEdit64.png
```

bigbigmdm, 2023-12-21:
The correct path is `/usr/lib/udev/rules.d/99-CH341.rules`.

bigbigmdm, 2023-12-21:
I changed IMSProg_programmer/CMakeLists.txt and rebuilt the deb package for UBUNTU 22.04 (Tag v1.1.2-12 and repository).

closed 2023-12-22 by mrv96