bigbigmdm/IMSProg #28: Error on Verify function

VittorioBurani, 2023-12-20:
`Verify` function on I2C EEPROMs fails when verifying files shorter than full EEPROM size because it searches for the dummy 0x00 used to pad the content by the `Program` function

Maybe adding some 0x00 padding also in the `Verify` function may solve the issue with hopefully low effort (in the virtual file content bytestring used to do the comparison you could pad accordingly to what happens during `Program`)

Issue find on Ubuntu 22.04 LTS, writing on the following IC:

- 24_EEPROM
- COMMON
- 24C256
- 32 K
- 32
- 3.3 V

bigbigmdm, 2023-12-21:
Thanks, I'll look into it.

bigbigmdm, 2023-12-25:
I fixed this error. Please test the new version.

closed 2023-12-25 by bigbigmdm

mrv96, 2023-12-28:
Issue not 100% fixed: https://github.com/bigbigmdm/IMSProg/pull/29#issuecomment-1871286177