bigbigmdm/IMSProg #31: QByteRef beyond QByteArray size

VasiliyTurchenko, 2023-12-25:
mainwindow.cpp (starting at line 617 ): 
```
    chipData = hexEdit->data();
    //uint8_t buf[currentBlockSize];
    uint8_t *buf;
    buf = (uint8_t *)malloc(currentBlockSize);
    for (k = 0; k < currentNumBlocks; k++)
      {

         for (j = 0; j < currentBlockSize; j++)
            {
               buf[addr + j - k * currentBlockSize] =  static_cast<uint8_t>(chipData[addr + j]) ;
            }
```
QByteArray ChipData initialized with the hexEdit->data(). But the user can load file of any size. If the file size is smaller than chip size, chipData will also have size < memory chip size.
Since writing algorithm alawys writes the full chip, and the chipData.size < currentNumBlocks*currentBlockSize, at some moment ```chipData[addr + j]``` will try to read bytes which are not persent in chipData. QT runtime warns about that with the message like "QByteRef beyond QByteArray size". Actually it is reading beyond array limits and this is undefined behaviour.

VasiliyTurchenko, 2023-12-25:
The mesaage is:
```
Using QByteRef with an index pointing outside the valid range of a QByteArray. The corresponding behavior is deprecated, and will be changed in a future version of Qt.
```

bigbigmdm, 2023-12-25:
I fix it.
    // if ChipSze = 0 IMSProg using at hexeditor only. chipsize -> hexedit.data
    // if ChipSize < FileSize - showing error message
    // if Filesize <= ChipSize - filling fileArray to hexedit.Data, the end of the array chipData remains filled 0xff
    QFile file(fileName);
    if ((info.size() > currentChipSize) && (currentChipSize != 0))
    {
      QMessageBox::about(this, tr("Error"), tr("The file size exceeds the chip size. Please select another chip or file or use block operations to split the file."));
      return;
    }
    if (!file.open(QIODevice::ReadOnly))
    {

        return;
    }
    buf.resize(info.size());
    buf = file.readAll();
    if (currentChipSize == 0)
    {
        chipData.resize(info.size());
    }

    for (uint32_t i=0; i < info.size(); i++)
    {
        chipData[i] = buf[i];
    }
    hexEdit->setData(chipData);

    file.close();

bigbigmdm, 2023-12-30:
Fixed

closed 2023-12-30 by bigbigmdm