bigbigmdm/IMSProg #37: 32Mb spi flash

VasiliyTurchenko, 2023-12-25:
IMSProg fails when trying to read 32Mb spi flash.
This happens because of  null pointer in spi_nor_flash.c
```
static int snor_4byte_mode(int enable)
{
	int retval;

	if (snor_wait_ready(1))
		return -1;

	if (spi_chip_info->id == 0x1) /* Spansion */
	{
```
The spi_chip_info does not exist.

bigbigmdm, 2023-12-25:
yes, the spi_chip_info does not exist. I need time to compare datasheets...

bigbigmdm, 2023-12-25:
After we get this program up and running, do you want to do CH347?

VasiliyTurchenko, 2023-12-25:
```
do you want to do CH347?
```
What do ou mean?
Everyhing has started after I wanted to fix a laptop and read its 32MB BIOS chip. I rememer IMSProg from Habr and I tried to use it, but it failed.
I started investigation why and yesterday have found unrelated issued and today I am moving to 4 byte addressing problem to finally get my BIOS )
Such a long way...

bigbigmdm, 2023-12-25:
Unfortunately I didn't have chips of that size on hand, hence the problem.

bigbigmdm, 2024-01-16:
The problem has been fixed in version 1.1.4.

closed 2024-01-18 by bigbigmdm