bigbigmdm/IMSProg #38: How to pack into .deb and .rpm files

mrv96, 2023-12-28:
Can you add to README.md the instructions to create .deb / .rpm after compilation?

bigbigmdm, 2023-12-28:
I'm not the one putting together the RPM packages. I can't help you there yet. I can write about DEB-packages, but I doubt that this information is needed in README.md. I have written in detail how to build with CMake. DEB packages are downloaded automatically when new versions are released from Launchpad repository.

P.S..
Creating a DEB package:
```
cd imsprog-1.1.3
dpkg-buildpackage -T clean

dh_make --createorig
debuild -S -us -uc -sa
dpkg-buildpackage -rfakeroot -us -uc
```

mrv96, 2023-12-28:
Thank you very much!

P.S. since you already use cmake, wouldn't it be easier to use cpack?

bigbigmdm, 2023-12-28:
One of my goals is to publish a package to Debian and Ubuntu. They have a number of strict requirements for all files. For example, I have redone the debian/control file about thirty times.

mrv96, 2023-12-28:
I know, it's quite tricky. When you have done it may be useful to have something like a Makefile target in order to create .deb simply with something like: `make package`

bigbigmdm, 2023-12-28:
Yes, you are right. I still need to add encryption key signatures to the files and send them to ubuntu/debian.

P.S. A utility to test the package:

lintian -i -I --show-overrides imsprog_1.1.3-3_amd64.deb
lintian -i -I --show-overrides imsprog_1.1.3-3-3.dsc
lintian -i -I --show-overrides imsprog_1.1.3.3-3_source.changes

bigbigmdm, 2023-12-30:
ok

closed 2023-12-30 by bigbigmdm

Fantu, 2024-01-23:
Small update for users that want to build deb package:
Go to debian/latest branch, the packaging for official build, it can be used also for manual build on some older Debian/Ubuntu.
For build binary on the current system simply do:
`dpkg-buildpackage -b -us -uc`
For build on clean environment outside (for example sending to launchpad or own build system:
`gbp buildpackage --git-builder='debuild -d -S -sa'`

bigbigmdm, 2024-01-24:
Thank you, @Fantu!