bigbigmdm/IMSProg #41: how to use tl866cs chip database in IMSprog?

lucasmz1, 2023-12-30:
hello dear friend, I have an tl866cs epron programmer and I want to use his database in your IMSprog, but I don't know how,
I saw in your repository that I can use the database of ezp2019-2025 too, can you give me a shortcut of how to add all the chip datbase from tl866cs in IMSprog, the one I have doesn't have .dat files to be added.

bigbigmdm, 2023-12-30:
Hello dear friend, I don't know where TL866 stores the chip data. If you show me the files where TL866 stores this data, there will be a topic for further work.

lucasmz1, 2023-12-30:
I belive it is in the dll file, I don't know for sure! here's the link of the download file if it might be interesitng for you.

http://www.autoelectric.cn/MiniPro/minipro_setup685.rar

lucasmz1, 2023-12-30:
have a look in David Griffith his is the one that ported tl866cs into linux.
https://github.com/radiomanV/TL866
https://gitlab.com/DavidGriffith/minipro
https://github.com/Kreeblah/XGecu_Software

bigbigmdm, 2023-12-30:
Be kind, please don't bombard me with links. Please find a file with data on microcircuits and specify the path to it.

lucasmz1, 2023-12-30:
ok sorry dear friend.

lucasmz1, 2023-12-30:
here it is https://gitlab.com/DavidGriffith/minipro/-/blob/master/infoic.xml?ref_type=heads

lucasmz1, 2023-12-30:
data on microcircuits and specify the path to it

> here it is https://gitlab.com/DavidGriffith/minipro/-/blob/master/infoic.xml?ref_type=heads

bigbigmdm, 2023-12-30:
Thank you! I'll see what I can do. Keep in mind, many series of chips cannot be connected to the CH341a. The TL866 is a more versatile device.

lucasmz1, 2023-12-30:
Thank you brother, I'll be around if you need something. see ya. now closing this issue.

closed 2023-12-30 by lucasmz1