bigbigmdm/IMSProg #55: README.md fix "What is this"

AndreiCherniaev, 2024-02-11:
### Description

What is different between [[EZP2019-EZP2025_chip_data_editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor)](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor) and [IMSProg editor](https://github.com/bigbigmdm/IMSProg/tree/main/IMSProg_editor)?

I want to fix this text
"IMSProg - Linux IMSProg - I2C, MicroWire and SPI EEPROM/Flash chip programmer for CH341A devices. The IMSProg is a free programmer tool based on [QhexEdit2](https://github.com/Simsys/qhexedit2) and modified [SNANDer programmer](https://github.com/McMCCRU/SNANDer).

This is a GUI program used widget QhexEditor. For setting the SPI chip parameters you can use the Detect button for reading chip parameters (JEDEC information reading) or manually setting it. The I2C and MicroWire EEPROM only manually selected."
Looks like this repo contains two different programs: programmer and editor. If so then you should describe the programmer and describe the editor...

P.S. Please use English or Russian.

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-02-11:
1. What is different between [EZP2019-EZP2025_chip_data_editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor) and [IMSProg editor](https://github.com/bigbigmdm/IMSProg/tree/main/IMSProg_editor)?
- only field names and folder search algorithm.
2. You're right. The editor should also be described, for example as (here)[https://antenna-dvb-t2.ru/IMSProg.php].

AndreiCherniaev, 2024-02-11:
> 1. What is different between [EZP2019-EZP2025_chip_data_editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor) and [IMSProg [...]

Я изучил ссылку. 
"Формат базы данных микросхем основан на формате программаторов EZP2019, EZP2020, EZP2023, Minpro I, XP866+. Для ее редактирования используется отдельный компонент программы."
Компонент программы - звучит очень непонятно.
Также вы говорите "Программа IMSProg". То есть в вашем мире есть программа IMSProg и она делится на два компонента: programmer, data base editor.

Я такого не встречал. Давайте рассмотрим какой-нибудь классический пример, да ещё и англоязычный. Если переводить на вашу разработку термины [из ](https://en.m.wikipedia.org/wiki/GNU_toolchain)
То получается что у вас есть 
collection of tools - набор утилит
1) Утилита programmer - programmer tool
2) Утилита data base editor - data base editor tool

Давайте примем такую терминологию?

bigbigmdm, 2024-02-11:
3. Есть еще [скрипт](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update) обновления базы. 

Кое-что я писал в man по каждому из компонентов.
Давайте применим эту терминологию, но programmer - главная утилита, может жить самостоятельно, а остальные лишь делают работу с programmer более комфортной.

AndreiCherniaev, 2024-02-12:
> 1. What is different between [EZP2019-EZP2025_chip_data_editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor) and [IMSProg editor](https://github.com/bigbigmdm/IMSProg/tree/main/IMSProg_editor)?
> 
> 
>     * only field names and folder search algorithm.

So users shouldn't use EZP2019-EZP2025_chip_data_editor because it is deprecated. Users should use IMSProg instead. Correct?

bigbigmdm, 2024-02-12:
@AndreiCherniaev EZP2019-EZP2025_chip_data_editor используется для других программаторов - (EZP2019 - EZP2025)[https://antenna-dvb-t2.ru/EZP_programmers.php]. У них только софт под Windows. Изначально я хотел написать программу под Linux для них, но у них закрытый код и его не заполучить. Мне понравился формат данных их базы микросхем и я использовал его в IMSProg, немного изменив "под себя" в процессе доработки программы.

bigbigmdm, 2024-02-12:
@AndreiCherniaev , я добавил информацию в README. Вы посмОтрите?

AndreiCherniaev, 2024-02-12:
> Кое-что я писал в man по каждому из компонентов. Давайте применим эту терминологию

Я вас не понимаю. Мы вчера кажется сошлись на том, что 
IMSProg is a collection of tools:
но вы пишите
"IMSProg consists of three executable modules:"
Поэтому я смущён...

bigbigmdm, 2024-02-12:
Поправил.

AndreiCherniaev, 2024-02-12:
> @AndreiCherniaev EZP2019-EZP2025_chip_data_editor используется для других программаторов - [EZP2019 - EZP2025](https://antenna-dvb-t2.ru/EZP_programmers.php). У них только софт под Windows. [...]

[image]
Can you add link https://antenna-dvb-t2.ru/EZP_programmers.php in github's "Website" field?
P.S. Please use "Qt" not "QT".

bigbigmdm, 2024-02-12:
В моем профиле есть [сайт](https://antenna-dvb-t2.ru), или Вы имеете ввиду в другом месте?

AndreiCherniaev, 2024-02-12:
> В моем профиле есть [сайт](https://antenna-dvb-t2.ru), или Вы имеете ввиду в другом месте?

Я про вот это место, сюда добавить https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor
Can you add link https://antenna-dvb-t2.ru/EZP_programmers.php in github's "Website" field?
[image]
P.S. Please use "Qt" not "QT".

bigbigmdm, 2024-02-12:
@AndreiCherniaev В первом приближении README готово. Можете посмотреть?

AndreiCherniaev, 2024-02-12:
> @AndreiCherniaev В первом приближении README готово. Можете посмотреть?

Ох у меня такая проблема что из-за незакрытого моего реквеста мои новые реквесты объединяются с незакрытым...
https://github.com/bigbigmdm/IMSProg/pull/57/commits/0542ae397e2019b71b6934a40cb8a63237755f6d

AndreiCherniaev, 2024-02-12:
> @AndreiCherniaev В первом приближении README готово. Можете посмотреть?

Сейчас думаю над этим текстом
**IMSProg** - Linux IMSProg - I2C, MicroWire and SPI EEPROM/Flash chip programmer for CH341A devices.

То есть IMSProg - Linux only? А почему так? Вроде Qt позволяет быть на всех платформах...

bigbigmdm, 2024-02-12:
@AndreiCherniaev Так, вроде, все слилось как надо?

AndreiCherniaev, 2024-02-12:
> @AndreiCherniaev Так, вроде, все слилось как надо?

yes

bigbigmdm, 2024-02-12:
@AndreiCherniaev "То есть IMSProg - Linux only? А почему так? Вроде Qt позволяет быть на всех платформах..." - под Windows это целая история с неподписанными драйверами на CH341A, под Linux они уже есть в ядре. Под MAC - не знаю...

bigbigmdm, 2024-02-12:
@AndreiCherniaev , как закончите с README, дайте знать. Я сделаю новый таг и выпущу release

AndreiCherniaev, 2024-02-12:
> @AndreiCherniaev , как закончите с README, дайте знать. Я сделаю новый таг и выпущу release

У меня такая идея. Уже имеются разделы "Chip database editor" и "Chip updater". А почему нет раздела "chip programmer"? Похоже надо сделать раздел "chip programmer" и перенести туда текст

For setting the SPI chip parameters you can use the Detect button for reading chip parameters (JEDEC information reading) or manually setting it. The I2C and MicroWire EEPROM only manually selected.

The chip database format is clone with EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers. You can edit the database use the EZP Chip data Editor

[image]

[image] 

А также всё что написано в разделе "## How to use:" до раздела "## Chip database editor"

Правилньо я думаю?

AndreiCherniaev, 2024-02-12:
> По сути перед "How to use:" нужно добавить Chip_programmer или IMSProg_programmer.

Допустим перед "How to use:" добавлю "IMSProg_programmer". А вот этот текст разве не нужно перенести в этот новый раздел "IMSProg_programmer"?

For setting the SPI chip parameters you can use the `Detect` button for reading chip parameters (JEDEC information reading) or manually setting it. The I2C and MicroWire EEPROM only manually selected.

The chip database format is clone with EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers. You can edit the database use the [EZP Chip data Editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor)

И вот эти пикчи перенести в новый раздел?
[image]

[image]

bigbigmdm, 2024-02-12:
Посмотрите, как сейчас: How to use --> chip programmer

AndreiCherniaev, 2024-02-12:
> Посмотрите, как сейчас: How to use --> chip programmer

Я считаю что "How to use:" вообще лишний... Но это не принципиально... С [текстом-то](https://github.com/bigbigmdm/IMSProg/issues/55#issuecomment-1938682151) что делать?

bigbigmdm, 2024-02-12:
"How to use:" я бы оставил для удобства навигации.

"For setting the SPI chip parameters you can use the Detect button for reading chip parameters (JEDEC information reading) or manually setting it. The I2C and MicroWire EEPROM only manually selected." - дубляж информации Chip type selecting - может совсем убрать?

"The chip database format is clone with EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers. You can edit the database use the [EZP Chip data Editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor)" - можно оставить
на старосм месте. По сути - информация по используемому формату.

AndreiCherniaev, 2024-02-12:
> "For setting the SPI chip parameters you can use the Detect button for reading chip parameters (JEDEC information reading) or manually setting it. The I2C and MicroWire EEPROM only manually selected." - дубляж информации Chip type selecting - может совсем убрать?

Согласен, убирайте!

> можно оставить на старосм месте.
Согласен!

bigbigmdm, 2024-02-13:
@AndreiCherniaev , мне кажется,это лишнее. Текст выше рассказывает об этом адаптере, на плате написано, что это такое.

AndreiCherniaev, 2024-02-13:
> @AndreiCherniaev , мне кажется,это лишнее. Текст выше рассказывает об этом адаптере, на плате написано, что это такое.

ok.

What about this images
`![CH341A black](img/ch341_black150.png)  [image] `
Может быть надо их подписать? К чему они вообще указаны?

bigbigmdm, 2024-02-13:
@AndreiCherniaev мы доредактировались с Вами. Из  шапки пропало, что это программа для для программатора CH341A.

bigbigmdm, 2024-02-13:
Это должно идти первым предложением.

AndreiCherniaev, 2024-02-13:
> Это должно идти первым предложением.

**IMSProg** - **I**2C, **M**icroWire and **S**PI EEPROM/Flash chip **Prog**rammer - is a program to read, write EEPROM chips using programmers based on CH341A.

Good?

bigbigmdm, 2024-02-13:
Good!

bigbigmdm, 2024-02-13:
Я поднял картинку и опустил фразу о форматах. Теперь понятно, для чего она.

AndreiCherniaev, 2024-02-13:
> Я поднял картинку и опустил фразу о форматах. Теперь понятно, для чего она.

Я могу только догадываться, что это
Example of programmers based on CH341A
Так?

AndreiCherniaev, 2024-02-13:
А вот эти картинки - это про что? Не надо ли их перенеси в "Chip programmer"?
```
![CH341A EEPROM programmer](img/IMSProg_1.png)
![CH341A EEPROM programmer](img/IMSProg.png) 
```

bigbigmdm, 2024-02-13:
Люди, которые этим занимаются знают эти картинки лучше нас с Вами. Это не существенно. И должны они быть сверху, чтобы сразу было видно о чем речь.

bigbigmdm, 2024-02-13:
Поверьте, эту программу на Linux ждали около 10 лет

AndreiCherniaev, 2024-02-13:
> Люди, которые этим занимаются знают эти картинки лучше нас с Вами. Это не существенно. И должны они быть сверху, чтобы сразу было видно о чем речь.

ok

bigbigmdm, 2024-02-13:
@AndreiCherniaev мне кажется, что у нас с Вами хорошо получилось. Еще бы man файлы посмотреть...

AndreiCherniaev, 2024-02-13:
> man файлы

Give link please...

bigbigmdm, 2024-02-13:
[1](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg.1.gz)
[2](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/IMSProg_editor.1.gz)
[3](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update.1.gz)

AndreiCherniaev, 2024-02-13:
> [1](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg.1.gz) [2](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/IMSProg_editor.1.gz) [3](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update.1.gz)

How I can make commit here?

bigbigmdm, 2024-02-13:
Download -> unzip -> edit ->zip->upload

AndreiCherniaev, 2024-02-13:
> Download -> unzip -> edit ->zip->upload It is a standard man document.

И как вы будете мои изменения отсматривать? Разжимать у себя обратно и сравнивать? How another software host man files? I believe should be better solution...

bigbigmdm, 2024-02-13:
Это одноразовая акция, так что да. Эти файлы, я надеюсь, меняться не будут.

AndreiCherniaev, 2024-02-13:
> Это одноразовая акция, так что да. Эти файлы, я надеюсь, меняться не будут.

Сначала я увижу 2 недостатка и поправлю. Через час ещё два. Завтра ещё. И каждый раз unzip-zip. Очень грустно. Надо бы узнать как другие проекты это автоматизируют и тоже автоматизировать?

bigbigmdm, 2024-02-13:
Это номинальные файлы, дающие краткую информацию и отсылающие к index.html. Чисто теоретически можно их жать с помощью [cmake](https://stackoverflow.com/questions/7050997/zip-files-using-cmake).

bigbigmdm, 2024-02-13:
@AndreiCherniaev просто lintian сообщил в предупреждениях о необходимости man инструкций, которые для GUI программы мало интересны, но традиция... Вот генерация Index.html меня сейчас занимает больше. Поскольку он создан на основе README.md и чтобы не тащить за собой три десятка картинок я вставил их внутрь документа в base64. Fantu нашел инструмент, который хорошо конвертирует [readme в html](https://github.com/bigbigmdm/IMSProg/issues/50), но изображения в bsd64 он не переводит.

bigbigmdm, 2024-02-14:
@AndreiCherniaev их нельзя исключать. Они копируются в систему при установке. Давайте пока оставим их в покое.

closed 2024-08-17 by bigbigmdm