bigbigmdm/IMSProg #58: drop qmake

AndreiCherniaev, 2024-02-12:
### Description

Looks like you use CMake and qmake both build system. But looks like you prefer [CMake](https://github.com/bigbigmdm/IMSProg/blob/main/build_all.sh). I think it is good choice!
Do you really need qmake? It will take a great power to keep up to date both. I suggest drop qmake (remove IMSProg_editor.pro and IMSProg.pro). 

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-02-12:
@AndreiCherniaev I only use CMake. The .pro files are only needed for my QT5 Creator. (Not used for builds).

bigbigmdm, 2024-02-12:
They are included in .gitignore. Since I use two computers in different locations for now I am comfortable storing .pro files on GitHub.

AndreiCherniaev, 2024-02-12:
> They are included in .gitignore. Since I use two computers in different locations for now I am comfortable storing .pro files on GitHub.

But .pro files no need to work in Qt Creator. 
File > Open file or project 
select 
CMakeLists.txt

bigbigmdm, 2024-02-12:
Yes, i thing it, but it is historical

AndreiCherniaev, 2024-02-12:
> Yes, i thing it, but it is historical

Если что я ваши проекты так и открываю в кьюти креэторе:
File > Open file or project
select
CMakeLists.txt

А из-за того что у вас .pro файлы лежат путаница возникнет, да и вы сами запутаетесь, когда у вас в креэторе проект собирается и работает одним образом, а через CMake собирается и работает другим образом. Или вообще не собирается... Настоятельно рекомендую избегать путаницы, то есть удалить .pro

bigbigmdm, 2024-07-02:
This is a secondary objective. Especially since I use these files when using QTCreator. Neither Debian nor RedHat Developers are embarrassed by their presence. There is a much more interesting task that I can't get to - adding support for the CH347T programmer. I need both strategic advice and help here. 
[image]
Today it can only detect the presence of the CH347 programmer...

AndreiCherniaev, 2024-07-03:
> Especially since I use these files when using QTCreator.

Please use CMakeLists.txt to open project in Qt Creator. If you will open CMakeLists.txt then Qt Creator will use CMake build system.
[image]
[image]

bigbigmdm, 2024-07-03:
I'm not convinced. I am not comfortable installing intermediate versions in /usr/bin. So I have to make two CMakeLists.txt - one for the release, one for myself. That's even more ugly.

AndreiCherniaev, 2025-03-14:
> [@AndreiCherniaev](https://github.com/AndreiCherniaev) , I'm not convinced. I am not comfortable installing intermediate versions in /usr/bin. So I have to make two CMakeLists.txt - one for the release, one for myself. That's even more ugly.

1) Why you need to installing any versions in /usr/bin? Why you need to install? When you are developer you can run directly from build/, isn't it?
2) if you really need to change settings try [cmake preset](https://habr.com/ru/companies/timeweb/articles/741868/)