bigbigmdm/IMSProg #66: Porting to macOS

sund3RRR, 2024-03-04:
### Description

IMSProg can be compiled on macOS natively and works. The only one problem is `udev` rules for device detection. MacOS doesn't have `udev` rules, so it is necessary to do it darwin-way (maybe using [diskutil](https://apple.stackexchange.com/questions/46951/is-there-a-mac-equivalent-for-udev-folder-on-linux)).  Unfortunately, as we discussed, @bigbigmdm couldn't do it himself because he doesn't know anything about macOS. It is really difficult to support multiple operating systems and it is normal.
I opened this issue to encourage people to help with porting IMSProg to macOS.
[image]

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-05-17:
@sund3RRRR, [this](https://wch-ic.com/downloads/CH341SER_MAC_ZIP.html) file won't help solve our problem?

sund3RRR, 2024-05-17:
> @sund3RRRR, [this](https://wch-ic.com/downloads/CH341SER_MAC_ZIP.html) file won't help solve our problem?

Вообще теоретически должно заработать
Не знаю как конкретно эта штука ставит драйвер, если она ставит расширения ядра, теоретически она может потребовать отключить SIP, если kext не подписан
Я вряд ли тут смогу помочь, потому что я сам не занимался системным программированием и на самом деле не очень разбираюсь как работает программатор, поэтому разбирался вместе с отцом :)
Проверю на досуге, как будет время

bigbigmdm, 2024-05-17:
Great, I'll be waiting!

sund3RRR, 2024-05-17:
В общем мне стало интересно и я поставил драйвер
К сожалению, он у меня не заработал. Почему-то macOS не видит девайс. Не знаю, что я не так сделал...

bigbigmdm, 2024-05-17:
Спасибо. Теперь знаем, что это не помогает...

asamahy, 2024-09-13:
i'm on Sonoma 14.6.1 and it compiles and OS identifies the CH341A programmer correctly however i am not able to verify if it works or not as the only chip i currently have access to isnt in the database `MX25L12835F`.
it does require `libusb` and `libftdi` to be installed.

[image]

```
$ ioreg -p IOUSB
+-o Root  <class IORegistryEntry, id 0x100000100, retain 26>
  +-o AppleUSBVHCIBCE Root Hub Simulation@80000000  <class AppleUSBRootHubDevic$
  | +-o Apple T2 Controller@80100000  <class AppleUSBDevice, id 0x100000533, re$
  | +-o Headset@80400000  <class AppleUSBDevice, id 0x100000537, registered, ma$
  | +-o Ambient Light Sensor@80300000  <class AppleUSBDevice, id 0x10000053c, r$
  | +-o FaceTime HD Camera (Built-in)@80200000  <class AppleUSBDevice, id 0x100$
  | +-o Touch Bar Backlight@80700000  <class AppleUSBDevice, id 0x100000546, re$
  | +-o Touch Bar Display@80600000  <class AppleUSBDevice, id 0x10000055e, regi$
  | +-o Apple Internal Keyboard / Trackpad@80500000  <class AppleUSBDevice, id $
  +-o AppleUSBXHCI Root Hub Simulation@01000000  <class AppleUSBRootHubDevice, $
  | +-o 4-Port USB 3.0 Hub@01100000  <class AppleUSBDevice, id 0x100004afb, reg$
  |   +-o USB 10/100/1000 LAN@01140000  <class AppleUSBDevice, id 0x100004b1e, $
  +-o AppleUSBXHCI Root Hub Simulation@14000000  <class AppleUSBRootHubDevice, $
    +-o 4-Port USB 2.0 Hub@14100000  <class AppleUSBDevice, id 0x100004b0b, reg$
      +-o USB UART-LPT@14110000  <class AppleUSBDevice, id 0x100004c4e, registe$
```

`flashrom` reads it fine though.
and the flashrom output
```
$ flashrom -V --programmer ch341a_spi -r temp.bin -c "MX25L12835F/MX25L12873F"
flashrom 1.4.0 on Darwin 23.6.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with LLVM Clang 15.0.0 (clang-1500.3.9.4), little endian
Command line (7 args): flashrom -V --programmer ch341a_spi -r 3bb.bin -c MX25L12835F/MX25L12873F
Initializing ch341a_spi programmer
libusb: info [darwin_detach_kernel_driver] no capture entitlements. may not be able to detach the kernel driver for this device
libusb: warning [darwin_detach_kernel_driver] USB device capture requires either an entitlement (com.apple.vm.device-access) or root privilege
Cannot detach the existing USB driver. Claiming the interface may fail. LIBUSB_ERROR_ACCESS
Device revision is 3.0.4
The following protocols are supported: SPI.
Probing for Macronix MX25L12835F/MX25L12873F, 16384 kB: compare_id: id1 0xc2, id2 0x2018
Added layout entry 00000000 - 00ffffff named complete flash
Found Macronix flash chip "MX25L12835F/MX25L12873F" (16384 kB, SPI) on ch341a_spi.
Chip status register is 0x00.
Chip status register: Status Register Write Disable (SRWD, SRP, ...) is not set
Chip status register: Bit 6 is not set
Chip status register: Block Protect 3 (BP3) is not set
Chip status register: Block Protect 2 (BP2) is not set
Chip status register: Block Protect 1 (BP1) is not set
Chip status register: Block Protect 0 (BP0) is not set
Chip status register: Write Enable Latch (WEL) is not set
Chip status register: Write In Progress (WIP/BUSY) is not set
This chip may contain one-time programmable memory. flashrom cannot read
and may never be able to write it, hence it may not be able to completely
clone the contents of this chip (see man page for details).
===
This flash part has status UNTESTED for operations: WP
The test status of this chip may have been updated in the latest development
version of flashrom. If you are running the latest development version,
please email a report to flashrom@flashrom.org if any of the above operations
work correctly for you with this flash chip. Please include the flashrom log
file for all operations you tested (see the man page for details), and mention
which mainboard or programmer you tested in the subject line.
You can also try to follow the instructions here:
https://www.flashrom.org/contrib_howtos/how_to_mark_chip_tested.html
Thanks for your help!
Skipping writeprotect-based unlocking for read/verify operations.
Reading flash... read_flash:  region (00000000..0xffffff) is readable, reading range (00000000..0xffffff).
```

bigbigmdm, 2024-09-14:
@asamahy , thanks for the good work!
Please press the `Detect` button in the IMSProg program. The program should detect MX25L12805D chip. This is the analog for MX25L12835F (Jedec ID - C22018).
If the chip is directly connected (no alligator clip is used and the chip has been unsoldered), IMSProg can read, erase and write this chip. Please press <Ctrl+P> then the `Read` button and post this screen shot of the registers here.

asamahy, 2024-09-15:
here you go
[image]

and i've verified IMSProg can read, erase and write successfully on macOS Sonoma 14.6.1.

asamahy, 2024-09-16:
and the Registers for `AT45DB041D`
[image]

only thing that did happen is even though the LSB is `1` which makes the page size 256 i had to change it myself as the database has it at 268 instead

bigbigmdm, 2024-09-16:
@asamahy , I'm about to finish renaming the chips in IMSProg and on the site from which the updates are downloaded and see why this is so...

bigbigmdm, 2024-09-16:
@asamahy , please reload and recompile IMSProg from the main branch and read AT45DB041 again. I am interested in the page size combobox value.

bigbigmdm, 2024-09-16:
@asamahy , please wait. I remember your pull request, I want to do everything consistently. I need to test your changes on at least two different linux systems.

asamahy, 2024-09-16:
yeah yeah take your time. i tested on ubuntu 22.04.

> @asamahy , please reload and recompile IMSProg from the main branch and read AT45DB041 again. I am interested in the page size combobox value.

i was testing this then to delete old version i got sidetracked and created an uninstall script then while pushing to a new branch my git client started bugging so i got sidetracked again trying to recreate what caused the bug. xD

anyways, your reply got me back on track and yes the page size is now correctly identified.

bigbigmdm, 2024-09-16:
@asamahy Thank you! "and yes the page size is now correctly identified." -  That's exactly what I was trying to achieve!

bigbigmdm, 2024-09-17:
Solved!

closed 2024-09-17 by bigbigmdm