bigbigmdm/IMSProg #69: add support for GIGADEVICE GD5FGQ4UB91G spi nand

spyke-ninja, 2024-04-09:
### Description

can you add the support for GD5FGQ4UB91G ? these family nand is supported in SNANDER

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-04-10:
@spyke-ninja, I have no plans to support NAND flash memory chips yet. 

For now, I want to add support for the AT45DBxxx series. 

Then I will try to add support for the CH347 high speed programmer. 

After that, we can think about NAND flash as well. Only there is a big problem - I don't have these chips in my home. I don't even know where they are used.

Kiwimannen, 2025-02-16:
> After that, we can think about NAND flash as well. Only there is a big problem - I don't have these chips in my home. I don't even know where they are used.

In some routers, Im trying to read a Winbond W25N01KV SPI NAND but have not found any program that supports it.

bigbigmdm, 2025-02-16:
@Kiwimannen , hello! 
Please refer to the console utility [snander](https://github.com/Droid-MAX/SNANDer). If it does not support your chip, please edit the file [spi_nand_flash.c](https://github.com/Droid-MAX/SNANDer/blob/master/src/spi_nand_flash.c) according to Datasheet.

Kiwimannen, 2025-02-16:
Thanks! In their list there is something very similar so it might work - and I will edit file.
Will try this immediately, thanks again :)

bigbigmdm, 2025-08-28:
@Kiwimannen , please build the 1.7.1 version in main branch? and test it!