bigbigmdm/IMSProg #70: Database format compatibility

AndreiCherniaev, 2024-04-09:
### Description

Does IMSProg use the same database format as EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers or similar? "Основан на формате" means "similar", but "is clone" sounds like "database format is the same"...

"Формат базы данных микросхем основан на формате программаторов EZP2019, EZP2020, EZP2023, Minpro I, XP866+."
https://antenna-dvb-t2.ru/IMSProg.php

"The chip database format is clone with EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers."
https://github.com/bigbigmdm/IMSProg/blob/main/README.md

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-04-10:
@AndreiCherniaev , until about Ver. 1.0.19 IMSProg and EZP programmers had the same format. Then I changed the assignment of fields 3B, 3E, 40 a byte. That is why I write "The chip database format is clone ...". Chip data Editor can still edit files for both types of programmers.

You can view the format of EZPxxx programmers [here](https://github.com/bigbigmdm/EZP2023_to_SNANDer).
You can view the IMSProg format [here](https://github.com/bigbigmdm/IMSProg?tab=readme-ov-file#chip-database-format).

AndreiCherniaev, 2024-04-10:
"The chip database format is clone ...".
Это значит что формат такой же. Верно? Но фактически ориентировочно с версии 1.0.19 это не так: форматы стали разные. Верно? Тогда надо убрать что "is clone"! Но и в любом случае фраза "The chip database format is clone ..." выглядит сомнительно с точки зрения английского, не так ли? Я не эксперт, но у вас кажется passive voice, то есть правильно было бы "is cloned"...

bigbigmdm, 2024-04-10:
Кажется, изначально там было "based on..."

bigbigmdm, 2024-04-10:
@AndreiCherniaev , пожалуйста, оцените последний вариант.

AndreiCherniaev, 2024-04-10:
Считаю, что нужно сразу указать, что форматы разные! Да, они похожи, да формат IMSProg был вдохновлён форматами EZP programmers, но это такие детали, которые едва ли кого будут интересовать при чтении readme...
Ru:
Формат базы данных IMSProg отличается от формата, используемого в программаторахc EZP2019, EZP2020, EZP2023, Minpro I, XP866+, чтобы редактировать базу данных в их формате, используйте (EZP Chip data Editor)[https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor?tab=readme-ov-file].

En:
The IMSProg chip database format is different from format used in EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers, to edit database in it's format use the (EZP Chip data Editor)[https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor?tab=readme-ov-file].

bigbigmdm, 2024-04-10:
@AndreiCherniaev , я бы написал по-другому:

"You can edit the database use the [EZP Chip data Editor](https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor)" - теперь вообще не нужная фраза. Я исторически ее перетаскиваю, а зря...

Я бы все-таки написал: "За основу формата базы данных IMSProg был взят и переработан (дополнен/изменен) формат базы данных программаторов EZP2019..."

AndreiCherniaev, 2024-04-10:
The format of the chip database is based on the format used in EZP2019, EZP2020, EZP2023, Minpro I, XP866+ (programmers)[https://github.com/bigbigmdm/EZP2019-EZP2025_chip_data_editor?tab=readme-ov-file]. Warning, formats is not the same!

bigbigmdm, 2024-04-10:
@AndreiCherniaev ,Превосходно!

AndreiCherniaev, 2024-04-11:
I see you have changed text to

The format of the chip database is based on the format used in EZP2019, EZP2020, EZP2023, Minpro I, XP866+ programmers. Warning, format is not the same!

You doesn't put URL (see my message above). In fact I don't should this link be or not... If not then let's close this issue.

bigbigmdm, 2024-04-11:
@AndreiCherniaev this link is unnecessary.

closed 2024-04-11 by AndreiCherniaev