bigbigmdm/IMSProg #72: Change github About

AndreiCherniaev, 2024-04-13:
### Description

I suggest new About text from [here](https://github.com/AndreiCherniaev/IMSProg_fix)

I make tags using [Wikipedia](https://en.wikipedia.org/wiki/Programmer_(hardware))...

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

AndreiCherniaev, 2024-04-19:
I am talking about this

[image]

AndreiCherniaev, 2024-04-19:
> @AndreiCherniaev Oooh, now I get it! Thanks!

I also suggest you to add tags:
chip-programmer 
prom-writer
! Like on my example...

bigbigmdm, 2024-04-22:
@AndreiCherniaev , I have once again supplemented the `readme.md` file. Please have a look at it.

AndreiCherniaev, 2024-04-23:
> @AndreiCherniaev , I have once again supplemented the `readme.md` file. Please have a look at it.

Thank you. Looks ok.

closed 2024-04-23 by AndreiCherniaev