bigbigmdm/IMSProg #73: systemd-dev by default

AndreiCherniaev, 2024-04-16:
### Description

README.md
"On Debian >=13 and Ubuntu >=23.10:
sudo apt-get install systemd-dev
"
Are you shure? I have Ubuntu 23.10 (but it was updated from previous Ubuntu 23.idon'tremember may be 04). systemd-dev already has installed... How we can check does systemd-dev already has installed by default on Ubuntu 23.10 and Ubuntu 24?

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-04-16:
@AndreiCherniaev , I don't think we need to change anything here. This was suggested by @Fantu. As a developer, he knows more about filling releases with packages than we do.

Fantu, 2024-04-16:
systemd-dev is not installed by default in clean minimal build environment (where all packages should be compiled)
to simplify the migration to the new package was added a dependency to it, that's why you would also see it in desktop installations, but it will later be removed and being a -dev package it is not installed by default in normal installations (which is why the split was done, to reduce development dependencies in most installations where they are not needed)

closed 2024-04-16 by AndreiCherniaev