bigbigmdm/IMSProg #80: Issue with translation

giovanni69, 2024-04-26:
### Distribution

Arch Linux

### Architecture

64 bit

### Kernel version

6.8.7 from Arch Linux

### IMSProg version

1.3.6

### Bug description

Open dialogSFDP.
The translation is perfect.
If you click on Read or on Write registers, part of the translation reverts to English (see Legend area).
I am sending the two pictures to give you a better idea.

[image]

[image]

bigbigmdm, 2024-04-26:
@giovanni69, Ok. Thank you! I just found a bug on this form - incorrect stretching of elements. Fixing it.

giovanni69, 2024-04-26:
Have you fixed the bug I reported?

bigbigmdm, 2024-04-26:
I just found it - `dialpgsfdp.cpp` line 54. I'll fix it tomorrow.

bigbigmdm, 2024-04-27:
@giovanni69 , I fixed it! Please test the new version.

giovanni69, 2024-04-27:
I will test it shortly and let you know.

giovanni69, 2024-04-27:
> @giovanni69 , I fixed it! Please test the new version.

OK, it works perfectly now. Bug fixed.

bigbigmdm, 2024-04-27:
@giovanni69 thanks for the quick testing! I'm publishing a new release with the tag v1.3.7.

giovanni69, 2024-04-27:
Yes, I noticed from your previous commits.

giovanni69, 2024-04-27:
You can close this issue.

bigbigmdm, 2024-04-27:
Ok

bigbigmdm, 2024-04-29:
@giovanni69 , I forgot to translate the bash script of chip database update. Please help me: I need to translate lines number 44 - 47 of the file [IMSProg_database_update](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update).

giovanni69, 2024-04-29:
Done!

bigbigmdm, 2024-04-29:
@giovanni69 , Thank you!

giovanni69, 2024-05-01:
I noticed that some fields in the editor are not translated:

Sector size
Type HEX
4 bit address

I am sending you a screenshot.

[image]

bigbigmdm, 2024-05-01:
@giovanni69 yes, they need to be translated.

giovanni69, 2024-05-01:
@bigbigmdm let me know which file contains those untranslated words.

bigbigmdm, 2024-05-01:
@giovanni69 , these words are in the file `/IMSProg_editor/ezp_chip_editor.cpp`, line numbers: 80 - 92 Translations are available in the file `/IMSProg_editor/language/chipEditor_it_IT.ts`.

giovanni69, 2024-05-01:
@bigbigmdm I just checked and those words had already been translated. I don't understand why they are not displayed in Italian.

bigbigmdm, 2024-05-01:
Oh, I'm sorry! I have two `issues` at once.
Maybe I found it: 
```
line 89 ezp_chip_editor.cpp -> "        <source>4 bits address</source>"
```
```
line 201 - 202 /chipEditor_it_IT.ts-> "         <source>4 bits
address</source>"
```
- non-equivalent strings. The 's' in the word `bit' is illegal.
I will update the translations tomorrow and ask you to translate again.

bigbigmdm, 2024-05-02:
@giovanni69 , I have updated the language files for chipEditor.

giovanni69, 2024-05-02:
@bigbigmdm Done!

bigbigmdm, 2024-05-02:
@giovanni69 , Thanks! I compiled also the .qm file.

giovanni69, 2024-05-02:
@bigbigmdm Yes, I have seen the commit. Thanks!

giovanni69, 2024-05-02:
@bigbigmdm can 'No' and 'Yes' that you see in this screenshot also be translated?

[image]

bigbigmdm, 2024-05-02:
@giovanni69 I am using the `Zenity` library to display this message. I don't know how I can translate these messages. In my language this dialogue also looks like `yes` and `no`.

bigbigmdm, 2024-05-02:
@giovanni69 This is interesting: `zenity --calendar` on my computer translated the names of these buttons.

giovanni69, 2024-05-02:
@bigbigmdm Unfortunately, I would not know how to help you. If you can translate those words, let me know so we can update the translation.

bigbigmdm, 2024-05-02:
@giovanni69 , Please delete the line number 64 and run the script!

giovanni69, 2024-05-02:
@bigbigmdm which script are you referring to?

bigbigmdm, 2024-05-02:
Yes

bigbigmdm, 2024-05-03:
@giovanni69 , good afternoon! I have added the appstream metadata files to the project:
- [imsprog.editor.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/imsprog.editor.appdata.xml)
- [imsprog.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.appdata.xml)
- [imsprog.database.update.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.database.update.appdata.xml)
Please check the translation in them.

giovanni69, 2024-05-03:
Done!

bigbigmdm, 2024-05-03:
@giovanni69 , thank you!

bigbigmdm, 2024-05-03:
@blackPantherOS , good afternoon! Since your participation in the project, the IMSProg programme has been further developed. Could you please re-correct the translations of the following files?

- [chipProgrammer_hu_HU.ts](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/language/chipProgrammer_hu_HU.ts)
- [chipEditor_hu_HU.ts](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/language/chipEditor_hu_HU.ts)
- [IMSProg.desktop](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg.desktop)
- [IMSProg_database_update.desktop](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update.desktop)
- [IMSProg_editor.desktop](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/IMSProg_editor.desktop)
- [IMSProg_database_update](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/IMSProg_database_update)
- [imsprog.editor.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/imsprog.editor.appdata.xml)
- [imsprog.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.appdata.xml)
- [imsprog.database.update.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.database.update.appdata.xml)

blackPantherOS, 2024-05-05:
> @blackPantherOS , good afternoon! Since your participation in the project, the IMSProg programme has been further developed. Could you please re-correct the translations of the following files?
>
> * [...]

I looked at the strings and didn't see an untranslated new rows these files.

bigbigmdm, 2024-05-05:
@blackPantherOS , sorry, I was inaccurate. All new phrases are automatic translations using the DEEPL online translator. These phrases may need translation corrections.