bigbigmdm/IMSProg #84: Issue with locale

Tian-HP, 2024-05-02:
### Distribution

Ubuntu 23.10

### Architecture

amd64

### Kernel version

Linux version 6.5.0-28-generic 

### IMSProg version

V 1.3.7

### Bug description

Chinese locale is zh_CN, not zn_CH. 
[image]

bigbigmdm, 2024-05-02:
@Tian-HP Thanks! I will rename the files. After that maybe you can help me fix the translation to your language?

Tian-HP, 2024-05-02:
Of course. My pleasure.

bigbigmdm, 2024-05-02:
@Tian-HP thanks! I fixed the wrong file names.

Tian-HP, 2024-05-02:
[Uploading IMSProg.patch.zip…]()

Tian-HP, 2024-05-02:
diff --git a/IMSProg_editor/IMSProg_editor.pro b/IMSProg_editor/IMSProg_editor.pro
index 38541f8..3d1c589 100644
--- a/IMSProg_editor/IMSProg_editor.pro
+++ b/IMSProg_editor/IMSProg_editor.pro
@@ -61,7 +61,7 @@ TRANSLATIONS += language/chipEditor_ru_RU.ts \
         language/chipEditor_es_ES.ts \
         language/chipEditor_hu_HU.ts \
         language/chipEditor_de_DE.ts \
-        language/chipEditor_zn_CH.ts \
+        language/chipEditor_zh_CN.ts \
         language/chipEditor_uk_UA.ts \
         language/chipEditor_pt_BR.ts \
         language/chipEditor_it_IT.ts
diff --git a/IMSProg_editor/language/chipEditor_zh_CN.ts b/IMSProg_editor/language/chipEditor_zh_CN.ts
index 85c5d2d..440e399 100644
--- a/IMSProg_editor/language/chipEditor_zh_CN.ts
+++ b/IMSProg_editor/language/chipEditor_zh_CN.ts
@@ -62,7 +62,7 @@
     <message>
         <location filename="../ezp_chip_editor.ui" line="150"/>
         <source>Del</source>
-        <translation>Del</translation>
+        <translation>删除</translation>
     </message>
     <message>
         <location filename="../ezp_chip_editor.ui" line="159"/>
@@ -97,7 +97,7 @@
     <message>
         <location filename="../ezp_chip_editor.ui" line="189"/>
         <source>Move down</source>
-        <translation>下移</translation>
+        <translation>向下移动</translation>
     </message>
     <message>
         <location filename="../ezp_chip_editor.ui" line="192"/>
@@ -153,7 +153,7 @@
     <message>
         <location filename="../ezp_chip_editor.cpp" line="81"/>
         <source>Manufacture</source>
-        <translation>制造</translation>
+        <translation>制造商</translation>
     </message>
     <message>
         <location filename="../ezp_chip_editor.cpp" line="82"/>
@@ -206,8 +206,8 @@ address</source>
         <location filename="../ezp_chip_editor.cpp" line="90"/>
         <source>Block
 size K</source>
-        <translation>Bloque
-tamaño K</translation>
+        <translation>块
+大小 K</translation>
     </message>
     <message>
         <location filename="../ezp_chip_editor.cpp" line="91"/>
diff --git a/IMSProg_programmer/IMSProg.pro b/IMSProg_programmer/IMSProg.pro
index e1708c0..db3e54c 100644
--- a/IMSProg_programmer/IMSProg.pro
+++ b/IMSProg_programmer/IMSProg.pro
@@ -95,7 +95,7 @@ FORMS += \
 TRANSLATIONS += language/chipProgrammer_ru_RU.ts \
          language/chipProgrammer_es_ES.ts \
          language/chipProgrammer_de_DE.ts \
-         language/chipProgrammer_zn_CH.ts \
+         language/chipProgrammer_zh_CN.ts \
          language/chipProgrammer_uk_UA.ts \
          language/chipProgrammer_hu_HU.ts \
          language/chipProgrammer_pt_BR.ts \
diff --git a/IMSProg_programmer/language/chipProgrammer_zh_CN.ts b/IMSProg_programmer/language/chipProgrammer_zh_CN.ts
index 0658858..863b020 100644
[... 221 lines cut ...]
     </message>
     <message>
         <location filename="../mainwindow.cpp" line="1138"/>
diff --git a/IMSProg_programmer/other/index.html b/IMSProg_programmer/other/index.html
index 7f89a7f..793e223 100644
--- a/IMSProg_programmer/other/index.html
+++ b/IMSProg_programmer/other/index.html
@@ -301,7 +301,7 @@ IMSProg_programmer/other/index.html                         /usr/share//doc/imsp
 IMSProg_programmer/language/chipProgrammer_ru_RU.qm         /usr/share/imsprog
 IMSProg_programmer/language/chipProgrammer_es_ES.qm         /usr/share/imsprog
 IMSProg_programmer/language/chipProgrammer_de_DE.qm         /usr/share/imsprog
-IMSProg_programmer/language/chipProgrammer_zn_CH.qm         /usr/share/imsprog
+IMSProg_programmer/language/chipProgrammer_zh_CN.qm         /usr/share/imsprog
 IMSProg_programmer/language/chipProgrammer_uk_UA.qm         /usr/share/imsprog
 IMSProg_programmer/language/chipProgrammer_hu_HU.qm         /usr/share/imsprog
 IMSProg_programmer/language/chipProgrammer_pt_BR.qm         /usr/share/imsprog
@@ -309,7 +309,7 @@ IMSProg_programmer/language/chipProgrammer_pt_BR.qm         /usr/share/imsprog
 IMSProg_editor/language/chipEditor_ru_RU.qm                 /usr/share/imsprog
 IMSProg_editor/language/chipEditor_de_DE.qm                 /usr/share/imsprog
 IMSProg_editor/language/chipEditor_es_ES.qm                 /usr/share/imsprog
-IMSProg_editor/language/chipEditor_zn_CH.qm                 /usr/share/imsprog
+IMSProg_editor/language/chipEditor_zh_CN.qm                 /usr/share/imsprog
 IMSProg_editor/language/chipEditor_uk_UA.qm                 /usr/share/imsprog
 IMSProg_editor/language/chipEditor_hu_HU.qm                 /usr/share/imsprog
 IMSProg_editor/language/chipEditor_pt_BR.qm                 /usr/share/imsprog

Tian-HP, 2024-05-02:
田红鹏
***@***.***

Tian-HP, 2024-05-02:
IMSProg.patch.zip

Tian-HP, 2024-05-02:
[IMSProg.patch.zip](https://github.com/bigbigmdm/IMSProg/files/15188983/IMSProg.patch.zip)

Tian-HP, 2024-05-02:
[IMSProg-2.patch.zip](https://github.com/bigbigmdm/IMSProg/files/15189456/IMSProg-2.patch.zip)

bigbigmdm, 2024-05-03:
@Tian-HP, please tell me:
1. Can I add your name to the translations section of the [readme.MD](https://github.com/bigbigmdm/IMSProg?tab=readme-ov-file#Translations) file?

2. Can I add your name in `debian/copyright` as the author of `chipEditor_zh_CN.ts` and `chipProgrammer_zh_CN.ts` files?

Tian-HP, 2024-05-03:
added

bigbigmdm, 2024-05-03:
@Tian-HP , good afternoon! I have added the appstream metadata files to the project:
- [imsprog.editor.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_editor/other/imsprog.editor.appdata.xml)
- [imsprog.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.appdata.xml)
- [imsprog.database.update.appdata.xml](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/other/imsprog.database.update.appdata.xml)

Please check the translation in them.

Tian-HP, 2024-05-03:
checked

bigbigmdm, 2025-02-18:
Hello @Tian-HP  , @giovanni69  , @blackPantherOS  ! I have prepared a new version of IMSProg with new form - SPI NOR FLASH security registers. I need help with the translation... Can you help me ?

bigbigmdm, 2025-02-18:
It is in 

[image]

blackPantherOS, 2025-02-18:
> It is in
> 
> [image]
> 
> The files are located at https://github.com/bigbigmdm/IMSProg/tree/main/IMSProg_programmer/language
> 
> DialogSecurity and Mainwindow sections.

"Aham, I see now! There are unfinished strings, not untranslated ones."

giovanni69, 2025-02-18:
> Hello [@Tian-HP](https://github.com/Tian-HP) , [@giovanni69](https://github.com/giovanni69) , [@blackPantherOS](https://github.com/blackPantherOS) ! I have prepared a new version of IMSProg with new form - SPI NOR FLASH security registers. I need help with the translation... Can you help me ?

I will do the translation as soon as I can

bigbigmdm, 2025-02-19:
Thank you, @giovanni69 !