bigbigmdm/IMSProg #96: странно работает...

orabeidm, 2024-05-12:
### Distribution

Debian 12

### Architecture

amd64

### Kernel version

6.1.0-21-amd64

### IMSProg version

1.3.8

### Bug description

первый раз сработало. но потом обращается на адрес такой
Couldn't open device 1a86:5512
хотя на самом деле программатор тут
Bus 002 Device 011: ID 1a86:5523 QinHeng Electronics CH341 in serial mode, usb to serial port converter

### Steps to reproduce

без понятия

### Expected behavior

нормальную работу программы

### Additional information

нету доп информации

bigbigmdm, 2024-05-13:
@orabeidm , для теста попробуйте `sudo IMSProg` в терминале. Если запустится, будем разбираться куда у Вас попал файл `99-CH341.rules`.

bigbigmdm, 2024-05-13:
@orabeidm , я только что протестировал версию v1.3.8 на Debian 12 Live CD. У меня все стабильно работает. Собирал командой `sudo ./build_all.sh`.

orabeidm, 2024-05-13:
удалил и его и файл 99-CH341.rules через который хотел айдишники  ранее переназначить принудительно. пересобрал как у вас указано. запускается, но ищет программатор так же не по тому адресу где он есть...

orabeidm, 2024-05-13:
да, всё верно, спасибо. я просто зелёный совсем, только вхожу в тему линукс и электроники.

closed 2024-05-13 by orabeidm