bigbigmdm/IMSProg #101: IMSProg programmer cmake need sudo

johndoe71rus, 2024-05-27:
### Distribution

Ubuntu 22.04

### Architecture

amd64

### Kernel version

6.5.0-35

### IMSProg version

commit 0573530

### Bug description

cmake ask sudo passwrd
```
~/build/IMSProg/IMSProg/IMSProg_programmer/build$ cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.25
-- Checking for module 'udev'
--   Found udev, version 249
[sudo] пароль для johndoe: 
```

### Expected behavior

why it ask?

bigbigmdm, 2024-05-27:
The sudo password is required to copy files to standard linux folders. Executable files must be copied to `/usr/bin`, icons to `/usr/share/icons`, translation files and chip database to `/usr/share/imsprog`, etc.

Please see the [project structure](https://github.com/bigbigmdm/IMSProg?tab=readme-ov-file#project-structure) section.

johndoe71rus, 2024-05-27:
ok. it need to install. but i just configure to build. 
Typically, the setup and build procedure does not require root rights. Version 1.3.7 not ask root

bigbigmdm, 2024-05-28:
Any version of the IMSProg need the root privileges. It is a standard CMake building commands (see [README.md](https://github.com/bigbigmdm/IMSProg/blob/main/README.md)):
```
mkdir  build
cd build
cmake ..
make
**sudo** make install
```

johndoe71rus, 2024-05-28:
```
Any version of IMSProg needs root privileges. 
...
Also, the udev module needs a sudo password  - Found udev, version 249
```
you're a little wrong
```
johndoe@test:~/build/msprog/IMSProg$  git status
HEAD detached at f7eb9bd
nothing to commit, working tree clean
johndoe@test:~/build/msprog/IMSProg$  cd ./IMSProg_programmer/build/
johndoe@test:~/build/msprog/IMSProg/IMSProg_programmer/build$  cmake ..
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Checking for module 'libusb-1.0>=1.0.20'
--   Found libusb-1.0, version 1.0.25
-- Checking for module 'udev'
--   Found udev, version 249
-- Configuring done
-- Generating done
-- Build files have been written to: /home/johndoe/build/msprog/IMSProg/IMSProg_programmer/build
johndoe@test:~/build/msprog/IMSProg/IMSProg_programmer/build$ make
[  3%] Automatic MOC and UIC for target IMSProg
[  3%] Built target IMSProg_autogen
[  6%] Automatic RCC for recource.qrc
[ 10%] Building CXX object CMakeFiles/IMSProg.dir/IMSProg_autogen/mocs_compilation.cpp.o
[ 13%] Building CXX object CMakeFiles/IMSProg.dir/main.cpp.o
[ 16%] Building CXX object CMakeFiles/IMSProg.dir/mainwindow.cpp.o
[ 20%] Building CXX object CMakeFiles/IMSProg.dir/qhexedit.cpp.o
[ 23%] Building CXX object CMakeFiles/IMSProg.dir/chunks.cpp.o
[ 26%] Building CXX object CMakeFiles/IMSProg.dir/commands.cpp.o
[ 30%] Building CXX object CMakeFiles/IMSProg.dir/dialogsp.cpp.o
[ 33%] Building CXX object CMakeFiles/IMSProg.dir/dialogrp.cpp.o
[ 36%] Building CXX object CMakeFiles/IMSProg.dir/dialogsfdp.cpp.o
[ 40%] Building CXX object CMakeFiles/IMSProg.dir/dialogsr.cpp.o
[ 43%] Building CXX object CMakeFiles/IMSProg.dir/dialogsetaddr.cpp.o
[ 46%] Building CXX object CMakeFiles/IMSProg.dir/searchdialog.cpp.o
[ 50%] Building CXX object CMakeFiles/IMSProg.dir/dialoginfo.cpp.o
[ 53%] Building C object CMakeFiles/IMSProg.dir/bitbang_microwire.c.o
[ 56%] Building C object CMakeFiles/IMSProg.dir/ch341a_gpio.c.o
[ 60%] Building C object CMakeFiles/IMSProg.dir/ch341a_i2c.c.o
[ 63%] Building C object CMakeFiles/IMSProg.dir/ch341a_spi.c.o
[ 66%] Building C object CMakeFiles/IMSProg.dir/flashcmd_api.c.o
[ 70%] Building C object CMakeFiles/IMSProg.dir/i2c_eeprom.c.o
[ 73%] Building C object CMakeFiles/IMSProg.dir/mw_eeprom.c.o
[ 76%] Building C object CMakeFiles/IMSProg.dir/spi_controller.c.o
[ 80%] Building C object CMakeFiles/IMSProg.dir/spi_eeprom.c.o
[ 83%] Building C object CMakeFiles/IMSProg.dir/spi_nand_flash.c.o
[ 86%] Building C object CMakeFiles/IMSProg.dir/spi_nor_flash.c.o
[ 90%] Building C object CMakeFiles/IMSProg.dir/timer.c.o
[ 93%] Building CXX object CMakeFiles/IMSProg.dir/dialogabout.cpp.o
[ 96%] Building CXX object CMakeFiles/IMSProg.dir/IMSProg_autogen/EWIEGA46WW/qrc_recource.cpp.o
[100%] Linking CXX executable IMSProg
[100%] Built target IMSProg
johndoe@test:~/build/msprog/IMSProg/IMSProg_programmer/build$
```
f7eb9bd is V1.3.7

bigbigmdm, 2024-05-28:
It's not a bug, it's a feature :)

johndoe71rus, 2024-05-28:
it's bad programming practice 
https://www.reddit.com/r/linux/comments/1ekd5w/why_is_it_dangerous_to_compilebuild_packages_as/?rdt=63359

bigbigmdm, 2024-05-28:
This is not a bad programming practice, but a requirement of Debian developers for compatibility between old and new distributions. udev is used to determine the requirement for the location of the rules file, which has changed in newer distributions. If you are not happy with this, remove lines 38 - 59 from [CMakeLists.txt](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/CMakeLists.txt).

Fantu, 2024-05-28:
The udev location is not a Debian specific requirement but is to detect correct udev location regardless of the distro use, I found it in another project, are you sure it really requires root?
Actual build system is not very good, have other issue and for example don't support uninstall (that is not good if build manually without packages), if anyone has knowledge and time to improve it that would be great. Unfortunately I don't have much time and even the few things I did take me a lot of research time using cmake.

johndoe71rus, 2024-05-28:
```
remove lines 38 - 59 from [CMakeLists.txt](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/CMakeLists.txt).
```
not work 
[image]

bigbigmdm, 2024-05-29:
It's very strange. I not have this problem:
[image]

bigbigmdm, 2024-05-29:
I comment the lines 39 - 58.

johndoe71rus, 2024-05-29:
I do not know why

Uploading imsprog_cmake-2024-05-29_10-54-22.mp4…

bigbigmdm, 2024-05-29:
Sorry - I tested in LinuxMint 21.2 - the result is fine (I posted it in the screenshot), but in Fedora 38 the result is the same as yours. I'll look into it...

Fantu, 2024-05-29:
Need to find a way to run cmake in verbose mode that shows all the commands it executes (like set -x in bash) and find which command sudo requires
probably this can help (not tested): https://stackoverflow.com/questions/57999508/verbose-output-for-cmake-command

johndoe71rus, 2024-05-29:
if it help 
[trace.txt](https://github.com/bigbigmdm/IMSProg/files/15481496/trace.txt)

johndoe71rus, 2024-05-29:
CMakeLists.txt(170):  execute_process(COMMAND bash -c sudo udevadm control --reload-rules

i coment all 
```
#Reloading USB UDEV rules
execute_process(
    COMMAND bash "-c" "sudo udevadm control --reload-rules" 
    OUTPUT_VARIABLE FOO
)
```
and cmake work fine.

bigbigmdm, 2024-05-29:
@johndoe71rus thank you for good work!

johndoe71rus, 2024-05-29:
I think need to move the 
#Reloading USB UDEV rules
to the make install stage. this is not needed at the assembly stage

bigbigmdm, 2024-05-29:
I find [here](https://cmake.org/cmake/help/latest/command/execute_process.html): "execute_process runs commands while CMake is configuring the project, prior to build system generation. Use the [add_custom_target()](https://cmake.org/cmake/help/latest/command/add_custom_target.html#command:add_custom_target) and [add_custom_command()](https://cmake.org/cmake/help/latest/command/add_custom_command.html#command:add_custom_command) commands to create custom commands that run at build time."

bigbigmdm, 2024-06-10:
I moved the procedure for updating the rules to `build_all.sh`.

closed 2024-06-10 by bigbigmdm