bigbigmdm/IMSProg #106: Support for the Winbond W25Q__JV family

wucke13, 2024-07-20:
### Description

I'd like to dump a [W25Q256JV](https://www.winbond.com/hq/product/code-storage-flash-memory/serial-nor-flash/index.html?__locale=en&partNo=W25Q256JV). A very similar model, the `W25Q256JW` [seems to be supported](https://github.com/bigbigmdm/IMSProg/blob/d21586397ab7fab819492fd438279de3439a0f6d/README.md?plain=1#L464).

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-07-21:
Hello, @wucke13 ! 
No, a very similar model is the W25Q256FV. 
The W25Q256JV and W25Q256FV have a 3.3V supply and JEDEC ID EF/40/19, and the W25Q256JW has a 1.8V supply and JEDEC ID EF/60/19. For programming, read and erase data, the W25Q256JV and W25Q256FV have equivalent commands and parameters.

bigbigmdm, 2024-08-05:
@wucke13 can I close this issue?

wucke13, 2024-08-05:
Yes please, thank you!

closed 2024-08-05 by wucke13