bigbigmdm/IMSProg #109: Added support MXIC 25L1606E

GermanAizek, 2024-08-01:
### Description

Datasheet: https://www.tme.eu/Document/90cf95a7114025302d33a68125e207ab/MX25L1606E.pdf

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-08-01:
@GermanAizek ,  MX25L1606E and MX25L1605D  chips have the same JEDEC ID - C2/20/15. They have the same read, erase, write operations for the programmer. Since IMSProg determines the chip by the JEDEC ID value, I do not add chips with the same JEDEC ID to the database on purpose. Otherwise the program will take data from the database of the first chip with such JEDEC ID value.

closed 2024-08-10 by bigbigmdm

GermanAizek, 2025-04-09:
@bigbigmdm 

> [@GermanAizek](https://github.com/GermanAizek) , MX25L1606E and MX25L1605D chips have the same JEDEC ID - C2/20/15. They have the same read, erase, write operations for the programmer. Since IMSProg [...]

No, Error reading block 0 with MX25L1605D

bigbigmdm, 2025-04-09:
Please see the `Chip information` - [ctrl] + p and Statur register 0.

If bits SPR0, BP4....BP0 are not zero, please set these bits to zero and read the chip again.

GermanAizek, 2025-04-09:
> Please see the `Chip information` - [ctrl] + p and Statur register 0.
> 
> If bits SPR0, BP4....BP0 are not zero, please set these bits to zero and read the chip again.

After read have Status Reg 0 all 1 and unique ID FFFFFFFFFFFFF, JEDEC also FF FF FF.

Flash memory itself is definitely alive because BIOS starts without any problems with it.

Editing the values does not help as it does not record them, there is no information in the terminal after clicking on "Write regs"

bigbigmdm, 2025-04-10:
You misunderstand me. All bits in register 0 must be set to 0.

GermanAizek, 2025-04-14:
> You misunderstand me. All bits in register 0 must be set to 0.

Nothing helped, without powering the board, the flash memory cannot be read at all, when the power supply is connected, the flash memory reads up to 3% and Error Reading Block 0, most likely it is necessary to solder SPI memory from board.

bigbigmdm, 2025-04-14:
I'm not even talking about that! 
The chip must necessarily be desoldered from the board!