bigbigmdm/IMSProg #116: Do not use zenity

DJs3000, 2024-10-22:
### Description

Михаил, здравствуйте. При запуске IMSProgupdater получаю ошибку
**/usr/bin/IMSProg_database_update: строка 63: zenity: команда не найдена**
Погуглил, это некий консольный "display dialog boxes", ну примерно понятно что это. Возможно ли избавиться от этой штуки? Там просто вереница зависимостей тянется за zenity и может быть есть возможность опционально не использовать это дабы не раздувать список зависимостей на +10 пакетов?

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-10-22:
Добрый день. Вообще странно. В большинстве разных Linux, которые я использовал zenity уже был в системе. Пока можете заменить просто на echo. Возможно, со временем, я перепишу IMSProg_database_update на C++, но это потребует дополнительных библиотек. Как минимум - Openssl-dev.

DJs3000, 2024-10-22:
Я добавил данный проект в Эльбрус ОС, а тут всё не так просто и любая зависимость добавляется ручками. Пока что zenity нет, а это гномовская софтина, которая тянет за собой часть гнома) openssl конечно же есть и с ним проблем нет.

DJs3000, 2024-10-22:
Замена на echo нормальный вариант. Не вижу большого смысла рисовать таблички ради вывода нескольких цифр, которые и так прекрасно читаются.

bigbigmdm, 2024-10-22:
Просто в графическом интерфейсе тогда придется запускать скрипт в терминале...

closed 2024-11-22 by DJs3000