bigbigmdm/IMSProg #118: Error reading block 0 in MANJARO/ARCH

bigbigmdm, 2024-11-12:
### Distribution

ARCH, Manjaro

### Architecture

amd64

### Kernel version

For example 5.15 from ubuntu package 5.15.0-56.62~20.04.1

### IMSProg version

1.4.4

### Bug description

When SPI NOR FLASH chips are used, an error message - ‘Error reading block 0’ - is triggered in the chip read procedure.
Command line output: ‘cb-out: error: LIBUSB_TRANSFER_TIMED_OUT
[image]
’

This might be - https://bbs.archlinux.org/viewtopic.php?id=298664

### Steps to reproduce

Запустите IMSProg, затем вставьте чип Winbond W25Q128 в программатор, нажмите кнопку чтения.

bigbigmdm, 2024-11-12:
[@kjkent](https://github.com/kjkent), can you connect to this problem?

kjkent, 2024-11-12:
I can certainly check, though these W25Q series chips are what I usually use imsprog for and haven't encountered an error I'd attribute to the program.

The libusb errors I've encountered have often been related to incorrect udev rules/permissions/another program accessing the device/what day of the week it is, it's pretty temperamental in my experience.

It might be a day or two until I have the time, but I will throw the closest chip I have into the ch341a and test. Will report back what I find.

I've been using NixOS for a bit, but will fire up an Arch installation for this. And when I get time, I'll see about packaging imsprog for NixOS.

bigbigmdm, 2024-11-13:
Thank you, @kjkent ! For NixOS, the package has already been created by [r-ryantm](https://github.com/NixOS/nixpkgs/commits?author=r-ryantm).

bigbigmdm, 2024-11-13:
@kjkent , i installed IMSProg from AUR repository on Manjaro 24.1.2 and I see this problem: "Error reading sector 0"...
I used Oracle virtualbox and downloaded the official release from Manjaro's website.

bigbigmdm, 2024-11-20:
This problem also is in [ArchLinux forum](https://bbs.archlinux.org/viewtopic.php?pid=2190843)

ItsPancho8, 2024-11-30:
happens on rpm as well, I cannot flash a fix vbios to my gpu because of this :v

bigbigmdm, 2024-11-30:
I don't understand what `rpm` is?

asamahy, 2024-12-01:
Maybe try with a macOS VM if you don't have access to a machine to rule out udev.

bigbigmdm, 2024-12-01:
@asamahy , I'm not going to rule out `udev`, as it works correctly on Linux systems. The question here is about incorrect operation of the program with the latest version of the Linux kernel or the latest versions of the Libusb-1.0 library.

onlylunix, 2024-12-17:
Void linux, libusb 1.0.27, there is no such error

bigbigmdm, 2024-12-18:
Thank you, @asamahy ! The Fedora 41 distribution also uses `Libusb 1.0.27` and this problem was also not detected.
Can you please tell me what your native language is? If it is not included in IMSProg, can you make a translation?

asamahy, 2024-12-18:
@bigbigmdm Arabic is my native language and I'll gladly translate if it isn't already done.

bigbigmdm, 2024-12-18:
Thank you, @asamahy ! 
A brief instruction is at the end of the `README.md` file...

Other than that, I now have another little project - [SFP-Master](https://github.com/bigbigmdm/SFP-Master)....

asamahy, 2024-12-18:
Ok great I'll take a look and start as soon as possible

bigbigmdm, 2024-12-18:
@asamahy Thank you so much again!

sinkapoy, 2025-02-07:
the same happens with Kubuntu 24.04 and the master's build

bigbigmdm, 2025-02-20:
Linux Mint 22.0 XFCE - the program did not work, LINUX MINT 22.1 XFCE - everything was fine!