bigbigmdm/IMSProg #119: add support for WINBOND 25Q80DVNIG

wucke13, 2024-12-09:
### Description

I'd like to read a `25Q80DVNIG`. The autodetect detects this as `W25Q80BL`, but each read I get different data back. I can not rule out interference in the target circuit to be the culprit.

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2024-12-10:
@wucke13 - if you use an alligator clip and a chip mounted on the motherboard, the reading result will be random! Please desoldering your chip and installed in programmer directly!
The 25Q80DVNIG and W25Q80BL chips have equivalent commands and parameters for read and write procedures.

@wucke13 , i need help to translate my new small project to Greman - [SFP-Master](https://github.com/bigbigmdm/SFP-Master/). Can you help me?

wucke13, 2025-01-05:
@bigbigmdm Took me some time to find a free spot, but here we go: https://github.com/bigbigmdm/SFP-Master/compare/main...wucke13:SFP-Master:main?expand=1 :smile:

closed 2025-01-05 by wucke13

bigbigmdm, 2025-02-20:
Hello @wucke13! I have updated [IMSProg](https://github.com/bigbigmdm/IMSProg). I have added the work with SPI NOR Flash security registers to the program. I need help to translate new phrases in [chipProgrammer_de_DE.ts](https://github.com/bigbigmdm/IMSProg/blob/main/IMSProg_programmer/language/chipProgrammer_de_DE.ts) file.