bigbigmdm/IMSProg #120: Add Support for changing I2C Transmission speed

extremeCrazyCoder, 2025-01-03:
### Description

Some EEPROMS do not support the default speed. A speed selection dropdown would be nice.

The speed of the connection can be changed by calling "ch341setstream" inside "ch341readEEPROM_param" (I added the call after the line "dprintf("Allocated USB transfer structures\n");" - however other positions might / should also work)

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2025-01-04:
@extremeCrazyCoder  Okay. I think I also need to send the speed value from the database to the form. Please don't close this issue until I have done that.

extremeCrazyCoder, 2025-01-04:
Thanks for the quick response

Just to be clear: I did not implement this in a production ready way. I just added the specific line for testing purposes. As of now I don't have much experience with GUI development.

What I forgot to mention is that there is a second place that the speed needs to be set in.  The "ch341writeEEPROM_param" function.

bigbigmdm, 2025-01-18:
This issue has been resolved in v1.4.5.

extremeCrazyCoder, 2025-01-19:
I tested it and it works quite good. However based on my testing the labels are wrong. It would be good if somebody else could validate this (maybe I got bad luck with the silicon lottery or my measuring Equipment produces wrong results).

Reading / Writing with the 20kHz option produce a frequency of 40kHz (read) and 42kHz (write)

[image]

[image]

100kHz -> 200kHz (read) / 210kHz (write)

[image]

[image]

400kHz -> 500kHz (read) / 516kHz (write)

[image]

[image]

750kHz -> 761kHz (read) / 761kHz (write)

[image]

[image]

I will close this issue now since the original issue is resolved and I am also not quite sure whether my measurements are accurate.

closed 2025-01-19 by extremeCrazyCoder