bigbigmdm/IMSProg #123: ITE i2c smbus read and write for IT8xxx ec

ndarjo, 2025-01-27:
### Description

hi, i watch somewhere in another github project (zephyrproject) + yt video + aliexpress. there are way to read or write EC of ITE ec IT8xxx series. there are source code / linux program written in C. the folder name is IT81202 Linux Flash Tool. you can found it in here https://www.ite.com.tw/en/product/cate2/IT81202 . the read and write use something like download board with IT8390 ec in the board. there are company which sell that (XZZ) using same programmer as the link i attach above. i am too dumb to learn the source code, but i guess, the IT8390 is only used for USB to i2c, uart, serial/parallel port. the bus speed is 400kbps i guess. i dont have donor board to test it(board with ITE ec), and change the VID and PID of the source code and change it with ch341a vid and pid and run the test.

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

ndarjo, 2025-01-27:
[image]

this is the ITE download board,

ndarjo, 2025-01-27:
https://github.com/c0d3z3r0/i2ite this project use F232H which supported in USBASp-flash, written in python

ndarjo, 2025-01-27:
there also iteflash.cc in here https://chromium.googlesource.com/chromiumos/platform/ec/+/HEAD/util

ndarjo, 2025-01-27:
How ITE EC reflashing works
An ITE EC is reflashed using a Servo by:
1. Sending special non-I2C waveforms over its I2C clock and data lines, to enable a debug mode / direct firmware update (DFU) mode.
2. Communicating with it using I2C, including transferring the actual EC image over I2C.  The ITE EC will only respond over I2C after receiving the special waveforms.

the special waveform is available in i2ite py file

ndarjo, 2025-01-27:
https://chromium.googlesource.com/chromiumos/platform/ec/+/refs/heads/main/docs/ite-ec-reflashing.md
this is how it work

ndarjo, 2025-01-27:
first thing how it work; the pc send a bit bangs waveform to the board with ITE ec to enable Direct Firmware Upgrade via SDA and SCL. i attach the what definition of that waveform. if the ch341a can send that bitbangs waveform, i guess ch341 can program the ITE ec,

[image]

bigbigmdm, 2025-01-28:
@ndarjo It's all good. What I can't figure out is, what do you want from me?

ndarjo, 2025-01-28:
what  i want from you is, can you integrate it into the imsprog?

bigbigmdm, 2025-01-28:
I don't think the general public would be interested in a very specific microcontroller. Besides, I don't have such a chip in stock. And without verification it is ineffective to write anything.

bigbigmdm, 2025-01-28:
I don't think the standard I2C interface will allow you to do this, but I think you can do it in GPIO mode - change level values of pins independently, as it is done in IMSProg and SNANDer to work with 93C46/93C56/93C66 chips... (MicroWire). In this case you will have to write your own I2C exchange... But the speed of operation will be very slow.

ndarjo, 2025-01-28:
yeah. sure. that what the official programmer of ITE chip done. it send non i2c signal in sda and scl to make the ITE ec chip get into direct firmware upgrade then accept the firmware we send via i2c protocol after we send the non i2c signal. the non i2c signal in scl is 100khz and in sda is 200khz. do you think it is possible to make the sda and scl pin in ch341a act as gpio and run at that frequency?

ndarjo, 2025-01-28:
[image]

this is that non i2c waveform

bigbigmdm, 2025-01-28:
Not as it works now in IMSProg. The communication speed is much slower.

ndarjo, 2025-01-28:
ah okay, actually the signal is only one way(the destination didnt need to respond back). from the pc with ch341 will send that waveform to the other mainboard with IT EC via i2c interface. so based on your experience, it is not possible to make that wavefrom in that frequency using ch341a? if it true, then i ican try to get one ft232h and try improve the code available.

bigbigmdm, 2025-01-28:
I think the ft232h has a lot more features than the CH341a. But it is more expensive and less popular....

ndarjo, 2025-01-28:
true. thats also my problem, couldnt buy ft232h in my country. thank you so much for answering. i will try to figure it out

GunayAnach, 2025-05-05:
I would be interested in this as well. Currently supported IT programmers starts from about 600AUD

bigbigmdm, 2025-05-05:
No, I haven't done that yet...

GunayAnach, 2025-05-05:
@bigbigmdm Happy to test on my end if you have spare time.

ndarjo, 2025-05-05:
https://github.com/c0d3z3r0/i2ite/issues/3 there are working code for ft232h in here. modify by codezero

stany-project, 2025-05-09:
Hola , estoy trabajando en lo mismo , por ejemplo su uso un RT809H, realizo una lectura , y sin desconectar nada coloco un esp32 , escaneo y aparecen mas direcciones i2c , como son las 0x79 la cual al leer esta el firmware del ITE y puede ser leido , pero sin usar el RT809h antes es imposible , el problema es emular con un esp32 el bitbaring necesario para poder poner el ITE en modo KING , mediante la secuencia que muestran las diferentes liberias , pero aun no lo he logrado.