bigbigmdm/IMSProg #132: IMSProg_database_update - ошибки при работе скрипта

r-a-sattarov, 2025-03-17:
### Distribution

OSL-9.1

### Architecture

e2k

### Kernel version

5.10

### IMSProg version

1.4.5

### Bug description

Если описать проблему кратко:

Открываем программу -
[image]

Получаем вывод - 
[image]

Проблема кроется в скрипте /usr/bin/IMSProg_database_update.

Проблемы:

1) Исполнение первого условия из 2-й строки всегда возвращает ноль, из-за этого некорректно выходим в ветку строки 66. А в блоке строки 66 как раз выполняется вызов окошка со второго скриншота.
Предлагаю первое условие видоизменить - создавать файл в случае отсутствия.

2) Даже если вручную создать файл, появляется окошко:
[image]

Непонятно, что происходит. А вот если запустить скрипт в консоли, то понятно:
```
/usr/bin/IMSProg_database_update 
--2025-03-17 14:39:33--  https://antenna-dvb-t2.ru/dl_all/IMSProg.Dat
Распознаётся antenna-dvb-t2.ru… ошибка: Временный сбой в разрешении имен.
wget: не удаётся разрешить адрес «antenna-dvb-t2.ru»
/usr/bin/IMSProg_database_update: строка 72: /root/.local/share/imsprog/IMSProg.Dat: Нет такого файла или каталога
cp: не удалось выполнить stat для '/tmp/IMSProg.Dat': Нет такого файла или каталога
/usr/bin/IMSProg_database_update: строка 77: /root/.local/share/imsprog/IMSProg.Dat: Нет такого файла или каталога
-1 -> -1
```
По-хорошему поставить проверку статуса возврата на wget и писать сообщение об ошибке.

3) Начиная с 66 строки отступов нет.
4) Начиная со строки 3 вместо кучи if-ов можно использовать case "$LANG" in и код сразу станет чище.

Прошу исправить проблемы скрипта.

bigbigmdm, 2025-03-17:
@r-a-sattarov , проверьте, пожалуйста. С моей точки зрения теперь все варианты отрабатываются корректно.

r-a-sattarov, 2025-03-18:
Да, теперь норм. Спасибо большое!

closed 2025-03-18 by r-a-sattarov