bigbigmdm/IMSProg #134: Add binary compare

bynect, 2025-03-28:
### Description

It would be very useful to have a sidebyside view of the binary that you modified. this way we can compare what we read and want to write.
If you like the idea I can try to add it myself

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2025-03-28:
Hello, @bynect ! I thought about this, but it would require the use of two HexEditor plugins. The main form would have a very large width. It's not pretty. There will be too much generally unnecessary information. I think you need a separate utility (or an additional `Binary compare` form) to compare two binary arrays, since the compare function is rarely needed.  Also, the difference in the values of the binary arrays should be highlighted in color. I will probably do this, but later.

bynect, 2025-03-28:
> Hello, @bynect ! I thought about this, but it would require the use of two HexEditor plugins. The main form would have a very large width. It's not pretty. There will be too much generally [...]

I was thinking of an optional view that pops out when needed. Also in my screen almost half of the screen is empty in the hex editor view. So I don't think the visual clutter is really a problem. My idea was something like binary diff tools, but without the need to save the dump and open another program.

bigbigmdm, 2025-04-16:
Hello, @bynect !
Have you thought about [this](https://github.com/bigbigmdm/Binary_file_comparator/tree/main) feature with the chip data and downloaded file?

bynect, 2025-04-16:
> Hello, @bynect !
> Have you thought about [this](https://github.com/bigbigmdm/Binary_file_comparator/tree/main) feature with the chip data and downloaded file?

Yes this looks very good. Similar to my idea 👍

bigbigmdm, 2025-05-16:
Hello, @bynect !
Done!

bynect, 2025-05-16:
> Hello, [@bynect](https://github.com/bynect) ! Done!

brilliant! I'll try asap

bigbigmdm, 2025-05-17:
Excuse me, @bynect! What is your native language? Maybe you can add a new language support to my IMSProg?

bynect, 2025-05-17:
> Excuse me, [@bynect](https://github.com/bynect)! What is your native language? Maybe you can add a new language support to my IMSProg?

I'm italian but I use english usually