bigbigmdm/IMSProg #135: Add support for CH347

Tarek-Hasan, 2025-04-01:
### Description

Please, add support for Ch347 - newer and high speed version ch341. [AsProgrammer](https://github.com/nofeletru/UsbAsp-flash); windows only programmer has support for this.

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2025-04-02:
Hello, @Tarek-Hasan !
I thought about it and built a test version of IMSProg. Now I can't get support for some chips - 45xxx, 93xxx and 25xxx (not SPI NOR). I will come back to this problem after some time. [Here](https://github.com/bigbigmdm/CH347_GUI_SPI_NOR_Flash_programmer) is a test version of an older program for SPI NOR FLASH only.

Tarek-Hasan, 2025-04-02:
[CH341A/CH347 Programmer](https://www.instructables.com/CH341A-Programmer/) includes two small tools - File Merger and File Slicer. It would be nice to have these utilities. [Source code](https://drive.google.com/drive/folders/17xf3EKIPe2Nhx2obE235PBRnkSKaU4uv) of the app.

[image]

[image]

bigbigmdm, 2025-04-02:
The `File Slicer` is not needed in the IMSProg, because IMSProg has  the `Save part` and `Load part` items in `Flle` menu. You can loading full image in buffer use `File->Load` command and then use `Flie->Save part` command for save piece of file to your computer. You can use `File->Load Part` to load many parts of a binary image...

Tarek-Hasan, 2025-05-06:
Your ch347 is based on abandoned ch347-nor-prog which is itself based on deprecated ch341prog by hackpascal. Instead, you should use, hackpascal's [Universal Flash Programmer](https://github.com/hackpascal/ufprog). Or, add support for CH347 in SNANDer.

bigbigmdm, 2025-05-06:
Thanks, @Tarek-Hasan ! 
 I'll definitely check out that repository!