bigbigmdm/IMSProg #140: Added support Winbond W25Q16CVS16

GermanAizek, 2025-04-09:
### Distribution

For example Debian 12, Ubuntu 22.04, Fedora 37 etc...

### Architecture

For example amd64, armhf, arm64 etc...

### Kernel version

For example 5.15 from ubuntu package 5.15.0-56.62~20.04.1

### IMSProg version

For example version 1.1.6 or commit c763e58

### Bug description

1

bigbigmdm, 2025-04-09:
Hello, @GermanAizek !
 Perhaps you meant `W25Q16CVSSIG`. It is compatible with the `W25Q16DV'.

GermanAizek, 2025-04-09:
> Hello, [@GermanAizek](https://github.com/GermanAizek) ! Perhaps you meant `W25Q16CVSSIG`. It is compatible with the `W25Q16DV'.

I've now figured out how to determine, I'm just comparing similarity bank sizes and voltage in unit.

bigbigmdm, 2025-04-09:
Yes. IMSProg compares the parameters with the JEDEC ID. Please see my [table](https://antenna-dvb-t2.ru/spi_nor_flash.php).

closed 2025-05-05 by bigbigmdm