bigbigmdm/IMSProg #141: bulid on macos M3

kurbatovmax, 2025-05-07:
### Distribution

bulid on macos M3

### Architecture

macos Apple M3 Max

### IMSProg version

 v1.5.3, origin/main, origin

### Bug description

when building for mac os, you will get this error

bash-3.2$ cmake ..
-- The C compiler identification is AppleClang 17.0.0.17000013
-- The CXX compiler identification is AppleClang 17.0.0.17000013
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /opt/homebrew/bin/pkg-config (found version "2.4.3")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.28
-- Configuring done (0.8s)
-- Generating done (0.1s)
-- Build files have been written to: /Users/kurbatov/devel/IMSProg/IMSProg_programmer/build

dialogsfdp.cpp:76:18: error: no matching function for call to 'SPI_CONTROLLER_Read_NByte'
   76 |         retval = SPI_CONTROLLER_Read_NByte(sfdpBuf.get(),3,SPI_CONTROLLER_SPEED_SINGLE);

IMSProg/IMSProg_programmer/dialogsfdp.cpp:83:27: error: type 'std::shared_ptr<uint8_t[]>' (aka 'shared_ptr<unsigned char[]>') does not provide a subscript operator
   83 |         jedecMan = sfdpBuf[0];
......
.....

bigbigmdm, 2025-05-07:
Hello, @kurbatovmax! This is very strange. These libraries are included in the header file. Please add the following lines to `dialogsfdp.cpp`:

```
extern "C" {
#include "ch341a_spi.h"
#include "spi_controller.h"
}
```
- after the line `#include <QDebug>`.

kurbatovmax, 2025-05-09:
> Hello, [@kurbatovmax](https://github.com/kurbatovmax)! This is very strange. These libraries are included in the header file. Please add the following lines to `dialogsfdp.cpp`:
> 
> ```
> extern "C" {
> #include "ch341a_spi.h"
> #include "spi_controller.h"
> }
> ```
> 
> * after the line `#include <QDebug>`.

thanks for the help, but I still get errors when building.

bash-3.2$ make 
[  0%] Built target IMSProg_autogen_timestamp_deps
[  2%] Automatic MOC and UIC for target IMSProg
[  2%] Built target IMSProg_autogen
[  4%] Building CXX object CMakeFiles/IMSProg.dir/mainwindow.cpp.o
/Users/maximk/devel/IMSProg/IMSProg_programmer/mainwindow.cpp:189:35: error: no matching constructor for initialization of 'std::shared_ptr<uint8_t[]>' (aka 'shared_ptr<unsigned char[]>')
  189 |        std::shared_ptr<uint8_t[]> buf(new uint8_t[step]);
      |                                   ^   ~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3654:23: note: candidate constructor not viable: no known conversion from 'uint8_t *' (aka 'unsigned char *') to 'nullptr_t' (aka 'std::nullptr_t') for 1st argument
 3654 |     _LIBCPP_CONSTEXPR shared_ptr(nullptr_t) _NOEXCEPT;
      |                       ^          ~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3668:5: note: candidate constructor not viable: no known conversion from 'uint8_t *' (aka 'unsigned char *') to 'const shared_ptr<unsigned char[]>' for 1st argument
 3668 |     shared_ptr(const shared_ptr& __r) _NOEXCEPT;
      |     ^          ~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3676:5: note: candidate constructor not viable: no known conversion from 'uint8_t *' (aka 'unsigned char *') to 'shared_ptr<unsigned char[]>' for 1st argument
 3676 |     shared_ptr(shared_ptr&& __r) _NOEXCEPT;
      |     ^          ~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3656:18: note: candidate template ignored: requirement 'is_convertible<unsigned char *, unsigned char (*)[]>::value' was not satisfied [with _Yp = uint8_t]
 3656 |         explicit shared_ptr(_Yp* __p,
      |                  ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3671:9: note: candidate template ignored: could not match 'shared_ptr<_Yp>' against 'uint8_t *' (aka 'unsigned char *')
 3671 |         shared_ptr(const shared_ptr<_Yp>& __r,
      |         ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3677:52: note: candidate template ignored: could not match 'shared_ptr<_Yp>' against 'uint8_t *' (aka 'unsigned char *')
 3677 |     template<class _Yp> _LIBCPP_INLINE_VISIBILITY  shared_ptr(shared_ptr<_Yp>&& __r,
      |                                                    ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3681:34: note: candidate template ignored: could not match 'weak_ptr<_Yp>' against 'uint8_t *' (aka 'unsigned char *')
 3681 |     template<class _Yp> explicit shared_ptr(const weak_ptr<_Yp>& __r,
      |                                  ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3696:9: note: candidate template ignored: could not match 'unique_ptr<_Yp, _Dp>' against 'uint8_t *' (aka 'unsigned char *')
 3696 |         shared_ptr(unique_ptr<_Yp, _Dp>&&,
      |         ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3705:9: note: candidate template ignored: could not match 'unique_ptr<_Yp, _Dp>' against 'uint8_t *' (aka 'unsigned char *')
 3705 |         shared_ptr(unique_ptr<_Yp, _Dp>&&,
      |         ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3652:23: note: candidate constructor not viable: requires 0 arguments, but 1 was provided
 3652 |     _LIBCPP_CONSTEXPR shared_ptr() _NOEXCEPT;
      |                       ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3664:26: note: candidate constructor template not viable: requires 2 arguments, but 1 was provided
 3664 |     template <class _Dp> shared_ptr(nullptr_t __p, _Dp __d);
      |                          ^          ~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3666:51: note: candidate constructor template not viable: requires 2 arguments, but 1 was provided
 3666 |     template<class _Yp> _LIBCPP_INLINE_VISIBILITY shared_ptr(const shared_ptr<_Yp>& __r, element_type* __p) _NOEXCEPT;
      |                                                   ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3659:9: note: candidate constructor template not viable: requires at least 2 arguments, but 1 was provided
 3659 |         shared_ptr(_Yp* __p, _Dp __d,
      |         ^          ~~~~~~~~~~~~~~~~~~
 3660 |                    typename enable_if<is_convertible<_Yp*, element_type*>::value, __nat>::type = __nat());
      |                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:3665:40: note: candidate constructor template not viable: requires 3 arguments, but 1 was provided
 3665 |     template <class _Dp, class _Alloc> shared_ptr(nullptr_t __p, _Dp __d, _Alloc __a);
      |                                        ^          ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
/Users/maximk/devel/IMSProg/IMSProg_programmer/snorcmd_api.h:33:5: note: candidate function not viable: cannot convert argument of incomplete type 'element_type *' (aka 'unsigned char (*)[]') to 'unsigned char *' for 1st argument; dereference the argument with *
   33 | int s95_write_param(unsigned char *buf, unsigned long to, unsigned long len, unsigned int sector_size, unsigned char currentAlgorithm);
      |     ^               ~~~~~~~~~~~~~~~~~~
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/IMSProg.dir/mainwindow.cpp.o] Error 1
make[1]: *** [CMakeFiles/IMSProg.dir/all] Error 2
make: *** [all] Error 2
bash-3.2$

bigbigmdm, 2025-05-09:
In my system (Linux mint 21) used 
```
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
```
You have the 
```
-- The C compiler identification is AppleClang 17.0.0.17000013
-- The CXX compiler identification is AppleClang 17.0.0.17000013
```
I realize the problem is here, but I don't know how to fix it today.....

asamahy, 2025-05-09:
@kurbatovmax please provide more information on your system. If you don't know how to then run this in Terminal:

```
system_profiler SPSoftwareDataType SPHardwareDataType |grep -e Chip -e "Model Name" -e "System Version"
```

kurbatovmax, 2025-05-10:
system_profiler SPSoftwareDataType SPHardwareDataType |grep -e Chip -e "Model Name" -e "System Version"
      System Version: macOS 15.4 (24E248)
      Model Name: MacBook Pro
      Chip: Apple M3 Max

> [@kurbatovmax](https://github.com/kurbatovmax), пожалуйста, предоставьте больше информации о вашей системе. Если вы не знаете, как это сделать, запустите это в Терминале:
> 
> ```
> system_profiler SPSoftwareDataType SPHardwareDataType |grep -e Chip -e "Model Name" -e "System Version"
> ```

system_profiler SPSoftwareDataType SPHardwareDataType |grep -e Chip -e "Model Name" -e "System Version"
      System Version: macOS 15.4 (24E248)
      Model Name: MacBook Pro
      Chip: Apple M3 Max

asamahy, 2025-05-14:
the projects build fine on Intel based mac running Sequoia 15.5.
i'm still trying to find a M machine so please bear with me.

kpbicka, 2025-05-28:
Собрал под MacOS, работает

[image]

сборка через жопу, конечно, много ругалось на Qt@5,  libusb, победить правильными методами не получилось, пришлось просто кинуть /opt/homebrew/Cellar/libusb/1.0.28/include/libusb-1.0/libusb.h в папку где происходит сборка (в build).

А ещё надо делать по шагам что написано внутри build.sh , либо исправлять его добавляя туда копирование libusb.h и исполнение последнего сmake из под sudo

Но всё получилось, 24c16 туда-сюда пишет-стирает (автоопределение конечно же не работает), с 25Qxx периодически вываливается ошибка

[image]

хз пока что это

bigbigmdm, 2025-05-28:
>  25Qxx периодически вываливается ошибка

- `24с16` и не должна автоматом определяться, т.к. это не предусмотрено самой микросхемой.
- С регистрами разбирались? [Здесь есть по русски с картинками](https://antenna-dvb-t2.ru/IMSProg.php).
- Прошиваете, случайно, не через прищепку?

asamahy, 2025-05-28:
это может быть проблемой от brew что он не добавляет qt5 в PATH

kpbicka, 2025-05-28:
А ещё сочетание  Cmd+Q вызывает Database Editor, хотя на Маках принято что Cmd+Q - это закрытие любого приложения (Quit). Ну и может шрифты как нибудь бы поправить на что-то более стандартное и покрупнее

jamartinezm, 2025-07-06:
I've modified the scrip to allow building with different revisions of libusb from homebrew and to fix the QT5 dependency issue:

```
#!/bin/bash

if [[ "$EUID" -ne 0 ]] && [[ "$OSTYPE" != "darwin"* ]]
  then echo "Please run as root! (sudo ./build_all.sh)"
  exit
fi
[[ "$OSTYPE" == "darwin"* ]] && export PATH="/opt/homebrew/opt/qt@5/bin:$PATH" && export C_INCLUDE_PATH=/opt/homebrew/opt/libusb:/opt/homebrew/include
(
cd IMSProg_programmer
rm -rf build/
mkdir build/
echo "STEP 1"
cmake -S . -B build/
echo "STEP 2"
cmake --build build/ --parallel
echo "STEP 3 - password needed"
sudo cmake --install build/
rm -rf build/
)
(
cd IMSProg_editor
rm -rf build/
mkdir build/
echo "STEP 4"
cmake -S . -B build/
echo "STEP 5"
cmake --build build/ --parallel
echo "STEP 6 - password needed"
sudo cmake --install build/
rm -rf build/
)
# Reloading the USB rules or creating the app bundles for macOS
[[ "$OSTYPE" != "darwin"* ]] && udevadm control --reload-rules || echo "STEP 7" && ./create_macos_appbundles.sh
```

jamartinezm, 2025-07-06:
You're correct that qt5 is missing on $PATH. It should be part of the installation steps (and honestly part of the cask).

The build script was ran on macOS Sequoia 10.5.
Disabling SIP was not needed and, for most people, it should always remain on.

Since the `cmake -install` command (that I have installed) will write to` /usr/local/bin/`, it will need elevated permissions on this macOS version. Can't speak for all the other versions or user configurations though.

I chose the easy way around this with sudo. None of the issues from #147  happened. Database update works fine, application launches without issues.

The correct way (that I can think of) would be to modify `CMakeLists.txt` by adding the `set(CMAKE_RUNTIME_OUTPUT_DIRECTORY "/new/path/`  option to an output folder that doesn't require elevated permissions.

The `create_macos_appbundles.sh` script would also need to be modified to point to the new output directory.

asamahy, 2025-07-07:
during the installation of homebrew it changes the permissions on `/usr/local/` to `USER:admin`, so yes unless you used homebrew to install the dependencies you'll need to `sudo` the install.

>The correct way (that I can think of) would be to modify ....
this might be a better way all together to avoid any permission conflict; however, it will break running from terminal unless added to `PATH` and i dont like editing `PATH` willy-nilly.

>The create_macos_appbundles.sh .....
the output is `/Applications` no need to change that, but if the location of the binaries is changed then that must be reflected in the app bundles.

are you running a Mx machine or Intel? @jamartinezm

bigbigmdm, 2025-08-26:
Не знаю, где Вы его скачиваете, в релизе с тегом 1.6.2 файл dialogabout.cpp содержит номер версии 1.6.2. Только что специально посмотрел. В ветке main лежит недоделанная версия 1.7.1 с поддержкой spi nand flash.

bigbigmdm, 2025-09-01:
[Здесь / Here](https://github.com/bigbigmdm/IMSProg/releases/tag/v1.6.2)