bigbigmdm/IMSProg #145: Support for FTDI FT2232H as interface

benryanau, 2025-06-06:
### Description

Fantastic work, I reckon this is headed to being the go-to for SPI on linux..

Without having reviewing the architechture, does IMSProg lend itself to extending support to another common (and better, with begtgter adapters eg Tigard) SPI interface IC?
The FTDI FT2232H is a good platform, flashrom supports it well (yes it can be tricky with the underlying mechanism/engine but I'd hope borrowing flashrom's module for it would be enough)

The CH341a has a lot of shortcomings in practice esp the lack of solid adapter hardware.. but there's a lot of very solid hardware based on the FTDI chip. There's others in the FTDI family that have more ports and flexibility too maybe easy enough to make it a universal FTDI module).

The Tigard seems to be the "best" platform to work from - no ZIF, so the pins do need to be jumper wired but with SOP16 so common 8 pin/ZIF is getting a bit obsolete)

Best of all it has 1.8v-5v or VTGT (periph driven Vcc) and but with the amount of SOP16's - no more piles of adapters on adapters. And an interposer board is getting necessary now to give caps/decoupling on Vcc as well as pullup/pulldowns for a few of the lines to get successful r/w.

Anyway thanks for the teriffic work!
Might help to clarify too what role your other projects have in relation to IMSProg too)
Ta

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2025-06-06:
Hello, @benryanau !
Today I am working on supporting SPI NAND chips. I don't know if I will succeed or not. SPI NAND communications are very complicated. In the future I plan to add support for CH347 programmer and FT232H programmer.

bigbigmdm, 2025-06-11:
I'm taking small steps. Very slowly. Last week, IMSProg could only read/write the NAND flash status registers :)

ndarjo, 2025-07-15:
> Hello, [@benryanau](https://github.com/benryanau) ! Today I am working on supporting SPI NAND chips. I don't know if I will succeed or not. SPI NAND communications are very complicated. In the future I plan to add support for CH347 programmer and FT232H programmer.

nice. i already ordered FT232H