bigbigmdm/IMSProg #146: Refactor database, fix issues with it.

LtqxWYEG, 2025-06-16:
### Description

I was writing this as a comment in another section and got really into it, until I figured out for myself how bad your database really is, and thought this has to be it's own issue. So, here it is. Hi :)

It's a bit of text, but it boils down to:  
- Please just throw out all old stuff and use the same XML based database as the other CH341 programmers do.  
- You are missing chips they have.  
- You don't have an entry for page-size in it.  
- The file is unreadable in text editor.  
- Duplicate JEDEC IDs are ignored - no pop-up or anything asking you to choose between differen variants after detection.  
- There is no ST manufacturer for 25xxxx chips. Initial user experience is bad if the chip isn't found in DB as there are only 4 manufacturer in 25_EEPROM list.
- and more, plese read if time. :)

---

Shouldn't every 25xxx be under "25_EEPROM"? I was wondering why this section is so empty. I also didn't find the manufacturer ST, but found that "MICRON" is having identical JEDEC IDs in your list. 
I Couldn't find my 25P05 chip in your database, saw that the 25_EEPROM section has only 4 manufacturer and gave up on your software, thinking the database is just really empty. However ...

Today I was in the mood of tinkering and had an issue with AsProgrammer, where a 25P05 has been reading a different set of clear text every time I read it, as if a function was called, and tried your program again. It was still version 1.5.3s database, so I updated it with the newest, thinking maybe now it will be detected. It wasn't. So I was about to give up on IMSProg again, but felt tinkery.
Looking manually, after the DB update, I ACCIDENTIALLY selected the "SPI FLASH" list, which is where I found a whole lot more than just 4 manufacturers. Thinking "hey, the new database is much better" I was looking through it for ST manufacturer but still couldn't find it. So I was thinking, "WTH? Why would there be no ST at all?" and looked into the database myself for anything close to "25p05" and its JEDEC ID and found that they are under "MICRON" for some reason, but just stop at 128K so that my chips weren't found. The 25P05 is the 64K variant of the 25P10

Reading through this issue, I can see that your database has a lot of problems. My own problems with it are proof of it. Is Micron really bigger than ST or are their chips more prevalent? What is going on here that you just don't have ST in your DB?

BTW, ST has A versions of the 25P05 and 25P10 with bigger page size. (You can find them in both AsProgrammers and the Original CH341s programmers DB.) Both of those are not in your DB:
```
      <ST25P05 id="202010" page="128" size="65536"/>
      <ST25P05A id="202010" page="256" size="65536"/>
      <ST25P10 id="202011" page="128" size="131072"/>
      <ST25P10A id="202011" page="256" size="131072"/>
```
My version of the 25P05 is an ST 25P05VP where I couldn't find a datasheet.  (Found at least a mention of the VP variant here, but no datasheet: https://datasheet.es/stock/price.php?item=25PO5VP) Seems they are 25P05 equivalents, not A, so max page size 128 when writing. I assume that is the "sector size" in your DB editor? Another issue! The only mention of "sectors" in the datasheet is here, and there are only two in the whole chip: "The memory is organized as 2 sectors, each containing 256 pages".  
The rst in that section reads: "Each page is 128 bytes wide. Thus, the whole memory can be viewed as consisting of 512 pages, or 65536 bytes." So, in your editor there is no "page size" even though it is important for it to be 128, not 256.

You know, you are allowed to just go through other DB and copy entries. (Or the sorting/naming scheme) Why didn't you do that?

Can your program handle duplicate JEDEC IDs in the DB? No it can't: There is no pop-up asking you to choose between variants of the same JEDEC ID.  
Then please add the non A variants of the 25P05 and 10 and maybe think about adding a whole 25xxx ST section.  
Actually, I did that to my DB and tried copying the new entries here for easy 5 sec implementation if you read my comment. But looking at the file in notepad, it seems to just not include the JEDEC ID, "sector"-size or anything else I can add in your editor.  
Seeing that, it would maybe just be better to redo the whole thing as a normal xml, like both the original CH341 programmer and AsProgrammer uses. Just much easier. (But that explains why you don't just go there and copy entries. You can't, without much effort)

You know, I make this a new issue. It's too bad and unreasonable.

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

hifron, 2025-11-29:
If there is maintained db like in datasheets, it exports(_if available or deals made_) could make it happen for some vendors if they maintain it. Manually refresh db in correct format is then possible via scripts, after that success also by github bot maybe on some trigger, but this should be better doing regularly and check diffs if they are correctly said. If this program is in Debian and maybe on Windows(_with windowsupdate drivers from officials, MS Store download or somewhere else(github pages ?)_), then maybe industry maintained db(_or something widely used_) could improve efficiency of UI suppliers like imsprog. Until then, it is only hobby project with try and error effort and maybe disasters of not only time waste. Industry of assemblers is waiting on keypress.

LtqxWYEG, 2025-12-10:
Don't understand what any of that has to do with the issue that the method of implementation of the database of chips is horrible for the user and should be remade to be one XML file like all the other programmer