bigbigmdm/IMSProg #148: Bad vertical alignment of the READ button

ipa64, 2025-07-07:
Mac Intel OS Sonoma.
IMSProg 1.6.2

I just noticed a small cosmetic detail, the READ button is a little lower than the DETECT button.
Maybe a difference between Qt on MacOS and Qt on Windows, because I haven't seen this on any other printscreen.
I noticed in the Mainwindows.ui file, that the READ button is contained in a QHBoxLayout which also includes a spacer. Even if you reduce the height of the spacer there is still a small vertical offset, probably a margin issue, I don't know, to test and work around the problem I included the same structure to the DETECT button, so the 2 buttons are aligned. 

[image]

[image]

bigbigmdm, 2026-10-20:
Could someone test the following UI file?

[mainwindow.ui.zip](https://github.com/user-attachments/files/33706946/mainwindow.ui.zip)

Please unpack this into the `/IMSProg_programmer/` folder.

s-em-i, 2026-10-20:
> 3\. Could someone test the following UI file?
>     [mainwindow.ui.zip](https://github.com/user-attachments/files/33706946/mainwindow.ui.zip)
>     Please unpack this into the `/IMSProg_programmer/` folder.

I replaced your my local file with the one you've provided and rebuild it via macport, and this is what I got!
[image]

bigbigmdm, 2026-10-20:
Hello, @s-em-i !
Could you also send me an image that isn't in full-screen mode, please?

s-em-i, 2026-10-20:
> 5\. Hello, [@s-em-i](https://github.com/s-em-i) !
>     Could you also send me an image that isn't in full-screen mode, please?
cant get it any smaller than this.
[image]

bigbigmdm, 2026-10-20:
@s-em-i Thank you! It's very good!

s-em-i, 2026-10-20:
The Vcc Volt Combobox doesnt seem to be allinged 100%.
Also in the bigger screen, this auto option window loses the line on the right which looks weird.

bigbigmdm, 2026-10-20:
> The Vcc Volt Combobox doesnt seem to be allinged 100%. 
 
- The form contains several hidden drop-down lists that appear when the chip type is changed... So, this is normal. You can change the `Type` to `24_EEPROM` or `SPI_NAND` for view it.

At the very least, it’s not as annoying as having buttons at different levels. But, in any case, this is something that will need to be revisited.

Thank you!