bigbigmdm/IMSProg #150: Support mx35uf4ge4ad-z4i

ystartgo, 2025-07-16:
### Description

Will Support mx35uf4ge4ad-z4i ?

### Usage tips

* Please use the 👍 [reaction](https://blog.github.com/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to show that you are interested into this.
* Please don't comment if you have no relevant information to add. It's just extra noise for everyone subscribed to this.
* Subscribe to receive notifications on status change and new comments, you can do also without add comment.

bigbigmdm, 2025-07-16:
How i write about you in readme and copyright files?

"- Tradiotional chinese translation by [ystartgo](https://github.com/ystartgo)" ?

bigbigmdm, 2025-07-31:
And erasing...